chipsalliance/f4pga-examples #19: Change to environment.yml

mithro, 2020-06-11:
### Current

```bash
INSTALL_DIR="/opt/symbiflow/xc7"
bash conda_installer.sh -b -p $INSTALL_DIR/conda && rm conda_installer.sh
source "$INSTALL_DIR/conda/etc/profile.d/conda.sh"
conda update -y -q conda

wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/4/20200416-002215/symbiflow-arch-defs-install-a321d9d9.tar.xz | tar -xJ -C $INSTALL_DIR
conda install -y -c symbiflow yosys yosys-plugins vtr-no-gui
conda install -y make lxml simplejson intervaltree git pip
conda activate
pip install python-constraint
pip install git+https://github.com/symbiflow/fasm
conda deactivate
```

### New version

```bash
wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh -O conda_installer.sh
bash conda_installer.sh -b -p $INSTALL_DIR/conda

source "$INSTALL_DIR/conda/etc/profile.d/conda.sh"
conda env update --file xc7.yml
conda activate
```

```yaml
name: xc7
channels:
- SymbiFlow
- defaults
dependencies:
- symbiflow
- yosys
- yosys-plugins
- vtr-no-gui
# Packages installed from PyPI
- pip:
  - -r file:requirements.txt
```

mithro, 2020-06-11:
FYI -- @pgielda @kgugala

mithro, 2020-06-11:
https://github.com/SymbiFlow/conda-env-make/blob/master/conda.mk is useful for understanding the steps.

pgielda, 2020-06-11:
Thanks. I am going to test this locally and report back.

mithro, 2020-06-11:
The steps are generally;

 1. Install conda
 2. Create environment from environment.yml + requirements.txt file
 3. Enter created environment

PR #24 (2020-06-22): Add conda environment file
Resolve #19
closed 2020-06-22 by kgugala