chipsalliance/f4pga-examples #25: The example directory seems useless?

mithro, 2020-06-19:
Currently we have
 - `symbiflow-examples/examples/xc7`

The extra `examples` directory doesn't seem to add much value?

It would probably be better to have the following
 - `symbiflow-examples/xc7`

It also means that a person who goes to this repo quickly sees which architectures there are examples for.

mithro, 2020-06-19:
@pgielda / @kgugala What do you think?

pgielda, 2020-06-19:
Not sure. The thing is that xc7, eos-s3 (and potentially other names in the future) will pile up that is one thing, second the idea with "examples" is to make clear that this is where the examples are (and not e.g. some arch data needed for the scripts, etc).

In any case I do not have any strong opinions about it in any case so we can as well remove it.

PR #47 (2020-09-11): Remove examples directory and improvement documentation
Fixes #25
Supersede of #39 

CI will fail, as tests are downloading master version, not version from this PR.

This pull request:

- Removes the extra examples directory.
- Makes the layout of the eos-s3 and xc7 directories consistent.
- Moves the setup instructions + build instructions under the relevant locations.
- Makes the setup instructions for eos-s3 and xc7 consistent (install conda then create an environment using the correct environment.yml file).
- Sections for the eos-s3 and xc7 are links to the README sections in each directo [... cut]