chipsalliance/f4pga-examples #26: Add an example for a Zynq part

mithro, 2020-06-25:
It appears we don't have any examples / demos which run on Zynq parts at the moment?

I don't know which board should be targeted?

mithro, 2020-06-25:
@kgugala -- Any thoughts?

mithro, 2020-06-25:
FYI - @andrewbutt1999

andrewb1999, 2020-06-25:
In my opinion, the important part of a zynq example is PS->PL communication.  After having 7020 support merged, it would be relatively easy to make an example based off of the zynq blinky example in prjxray.  I have a few tests that are more complicated, such as using axi, but they currently use xilinx IP.

At a more fundamental level, examples using the PS will be somewhat limited at the moment due a lack of understanding about configuring the PS using assembly.  For my more complicated examples I use the first stage boot loader generated by Xilinx, which contains intial PS configuration, and run the test from an SD card using antmicro's zynq-mkbootimage.

Eventually I think it will be necessary to fuzz the PS configuration (by checking different options in Vivado and comparing the resulting fsbl), but I'm not sure of the intellectual property concerns of using a Vivado/SDSoC/Vitis generated fsbl for now.

mithro, 2020-06-25:
I think Antmicro have a bunch of examples with that?

kgugala, 2020-06-25:
We have this one https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc/xc7/tests/ps7/axi_lite_reg it implements an AXI peripheral which can be then used with this tester https://github.com/SymbiFlow/symbiflow-xc7z-automatic-tester

I also implemented a LiteX-Linux design using PS7 DDR in Zynq. I have to find it, though. I'll link it here once I have it.

andrewb1999, 2020-06-25:
Oh ok that's interesting.  Does using U-boot replace using the fsbl?  Do you have any examples of running a bare metal application on the PS?

antoinevg, 2020-06-25:
I've got a [Zybo Z7](https://reference.digilentinc.com/reference/programmable-logic/zybo-z7/start) with a `xc7z020clg400-1` here and would be happy to help out with any testing.

antoinevg, 2020-06-25:
> Oh ok that's interesting. Does using U-boot replace using the fsbl? Do you have any examples of running a bare metal application on the PS?

I've been using this as a base: https://github.com/SymbiFlow/prjxray/tree/master/minitests/z7_blinky_emio

kgugala, 2020-06-25:
> Oh ok that's interesting. Does using U-boot replace using the fsbl? Do you have any examples of running a bare metal application on the PS?

fsbl simply configures the low level part of the chip (including DDR controller), optionally programs the FPGA and loads the next stage software. Instead of FSBL you can use U-Boot SPL but you need to ensure that is has a correct chip configuration for the board you're using. What you can do, is to export ps7_init_gpl.{c|h} from Vivado and use them with U-Boot SPL.

Note that mainline U-Boot has configurations for the most popular Zynq devboards.

Also you can integrate the init code with your app (as in blinky emio case). You need to remember about a few things:

* On startup PS in Zynq will it's internal bootloader which will load the next stage software to internal chip memory from a boot device.
* Boot device is determined by reading pin status (on e.g. Zybo you can set those with a jumper). 
* The software has to be small enough to fit the internal chip's memory (256 KiB)
* The software must be packed in boot image - you may use https://github.com/antmicro/zynq-mkbootimage to pack your application

PeterHindes, 2021-06-21:
I have a pynq z2 board with a zynq 7020 if you want some testing. I'm a bit of a noob though lol.

IanBoyanZhang, 2021-09-08:
Hello, I know this is an old thread. I recently got a pynq z1 board. Would love to learn about Symbiflow/yosys workflow (eventually being able to contribute). Wonder if it is a right board for a starter. Thanks!

pgielda, 2021-09-09:
Pynq Z1 has an XC7Z020 FPGA chip so I see no reason why in principle it would not work.
I remember that Pynq Z1 was somehow convoluted in terms of good documentation on how things are connected around and had no U-Boot port (but maybe I am mistaken?) so you might have some problems on this side (not strictly related to Symbiflow toolchain itself). 

It definitelty would be an interesting experiment to get Pynq working.

Asoftyn, 2022-12-03:
Hi, im new to the whole FPGA world and i just got myself an Pynq Z2, I wonder what i would need to do to get the bare minimum up and started. LED blink etc. 
Any help would be really appreciated.
Sorry if this is the wrong place to ask for advice on.

IanBoyanZhang, 2022-12-04:
I ended up using a Zedboard board. I am familiar with Python, but the whole PYNQ concept really confused me at the beginning. I had hard time to understand what happens at which abstraction layer. If you are new to both FPGA world and the Pynq board. I would say staying with the official toolchain and samples for now. Good thing about Symbiflow and Yosys is one wouldn't need a board to start learning about the toolchain.

Asoftyn, 2022-12-04:
The official toolchain is fine, but if i where to get started with using the f4pga toolchain what whould be a good place to start? I want to write my own example like blinking LED etc for the pynq z2. I think there should be support already for the chip aswell. How would i go about to start using the toolchain? What docs would be good to read to get started? I looked around a little bit but im a little confused.

mithro, 2022-12-06:
There was support for the Zybo Z7 in F4PGA examples but they seem to have disappeared? The using the board section still exists -> https://f4pga-examples.readthedocs.io/en/latest/running-examples.html#zybo-z7-board

Asoftyn, 2022-12-07:
Thank you, im looking around inside arch-defs and i can find support for the xc7z020 chip, how do i add a board for arch-defs and a test case for it? or how can i find appropiate F4PGA_PACKAGES packages for my board in the f4pga-examples so i can tweak the example package and scripts to run for my zynq board.

umarcor, 2022-12-07:
In 'Building example designs' > 'Xilinx 7-Series' there is a list of examples: https://f4pga-examples.readthedocs.io/en/latest/building-examples.html#xilinx-7-series.
The 'Counter test' example includes guidelines for Zybo Z7: https://f4pga-examples.readthedocs.io/en/latest/xc7/counter_test.html.
There is a gif of the board running the example: https://f4pga-examples.readthedocs.io/en/latest/_images/counter-example-zyboz7.gif.

@Asoftyn, in order to adapt the example, I guess you'd need to:

- Add an XDC file.
- Edit https://github.com/chipsalliance/f4pga-examples/blob/main/xc7/counter_test/Makefile#L5-L23 to set the XDC file for the pynq_z2 target.
  - Update https://github.com/chipsalliance/f4pga-examples/tree/main/xc7/counter_test accordingly.
- Edit https://github.com/chipsalliance/f4pga-examples/blob/main/common/common.mk#L7-L40 to set the device, part name, etc. for the pynq_z2 target.
  - Note that pynq_z2 is supported by openFPGALoader (OFL) already: https://trabucayre.github.io/openFPGALoader/compatibility/board.html#compatibility-boards.
  
As you saw, although 7z020 is found in arch-defs, (see https://github.com/f4pga/f4pga-arch-defs/tree/main/xilinx/xc7/archs/zynq7_z020), it seems that a different packages is available only: https://github.com/f4pga/f4pga-arch-defs/blob/main/xilinx/xc7/archs/zynq7_z020/devices/CMakeLists.txt. Actually, f4pga/f4pga-arch-defs#1260 included `xc7z020clg400-1`, but it was closed, not merged. /cc @tmichalak @kgugala

Hence, I recommend to open an issue in arch-defs in order to add the missing package to the definition artifacts/assets (which are then to be used in this repo). Related: f4pga/f4pga-arch-defs#2737.

When the arch-defs artifacts/assets are updated, either test the "Latest" workflow in this repo or edit the F4PGA_TIMESTAMP and F4PGA_HASH in https://f4pga-examples.readthedocs.io/en/latest/getting.html#setup-and-download-assets.

Asoftyn, 2022-12-07:
Thank you for taking the time to help me,  I will try and do this and then update with my progress, when I succeed I'll share what I've done so others can use it as well.
Thank you

tmichalak, 2022-12-14:
@Asoftyn The support for Zynq 7020 has been added to arch-defs in [f4pga/f4pga-arch-defs#1506](https://github.com/f4pga/f4pga-arch-defs/pull/1506) from @andrewb1999. Hence, to use it in f4pga-examples you could add a target for pynq to the https://github.com/chipsalliance/f4pga-examples/blob/main/common/common.mk#L7-L40 file as suggested by @umarcor

Asoftyn, 2022-12-15:
I got the `counter_test` example to work for pynq z2.

First during the Getting F4PGA steps i changed a few small things, in the `F4PGA_PACKAGES` variable i added `xc7z020_test` package so for PYNQ_Z2 it would be `F4PGA_PACKAGES='install-xc7 xc7z020_test'`
Then i changed the package install script for latest packages with the following: 
```bash
mkdir -p $F4PGA_INSTALL_DIR/$FPGA_FAM

for PKG in $F4PGA_PACKAGES; do
wget -qO- $(wget -qO- https://github.com/SymbiFlow/f4pga-arch-defs/releases/download/latest/symbiflow-${PKG}-latest) \
      | tar -xJC ${F4PGA_INSTALL_DIR}/${FPGA_FAM}
done
```

Now inside the F4PGA_EXAMPLES counter_test i added the following:
in `common/common.mk` added to the end (if this isnt correct please tell me)
```
else ifeq ($(TARGET), pynq_z2)
  DEVICE := xc7z020_test
  BITSTREAM_DEVICE := zynq7
  PARTNAME := xc7z020clg400-1
  OFL_BOARD := $(TARGET)
else
```

In `xc7/counter_test/Makefile` i added 
```make
else ifeq ($(TARGET),pynq_z2)
  XDC := ${current_dir}/pynq_z2.xdc
endif
```
Added the file `pynq_z2.xdc`

```
set_property -dict { PACKAGE_PIN H16   IOSTANDARD LVCMOS33 } [get_ports { clk }]; #IO_L13P_T2_MRCC_35 Sch=sysclk
create_clock -add -name sys_clk_pin -period 8.00 -waveform {0 4} [get_ports { clk }];

set_property -dict { PACKAGE_PIN R14   IOSTANDARD LVCMOS33 } [get_ports { led[0] }]; #IO_L6N_T0_VREF_34 Sch=led[0]
set_property -dict { PACKAGE_PIN P14   IOSTANDARD LVCMOS33 } [get_ports { led[1] }]; #IO_L6P_T0_34 Sch=led[1]
set_property -dict { PACKAGE_PIN N16   IOSTANDARD LVCMOS33 } [get_ports { led[2] }]; #IO_L21N_T3_DQS_AD14N_35 Sch=led[2]
set_property -dict { PACKAGE_PIN M14   IOSTANDARD LVCMOS33 } [get_ports { led[3] }]; #IO_L23P_T3_35 Sch=led[3]
```

Now entering the conda enviroment according to the F4PGA_EXAMPLES i was able to build it using `TARGET="pynq_z2" make -C counter_test` and then upload it to my pynq z2 board using `TARGET="pynq_z2" make download -C counter_test`

And my board is blinking nice and beautifully. Thank you all for the guidance and help.

tmichalak, 2022-12-15:
@Asoftyn that's great. Since you have already done and tested the modifications feel free to open a PR.

Asoftyn, 2023-01-02:
Im not really sure how to make a PR on github and i dont wanna screw things up

pgielda, 2023-01-02:
You can look at other PRs. Basically just create your own fork, create a branch with your changes and open a PR from this branch. 
Github describes this here: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork