chipsalliance/f4pga-examples #27: Add support for the 100T variant of the Arty A7

mithro, 2020-06-25:
It would be good if the examples also worked on the Arty 100T version.

mithro, 2020-06-25:
@tcal-x -- Any idea if this would be possible yet or are there things which need to be done first?

tcal-x, 2020-06-25:
Should just work.
But let me check :)

tcal-x, 2020-06-25:
Oh, wait, I see we need to manually pick what files are packaged, and the 100t files are not currently included under /opt/symbiflow/xc7/install/share/arch/.   The first needed file that the tool can't find is the arch.timing.xml file.

tcal-x, 2020-06-26:
Maybe we just need to grab a more recent archive/artifact.   The instructions tell you to `wget` this:
```
https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/4/20200416-002215/symbiflow-arch-defs-install-a321d9d9.tar.xz
```
from April 16.

@kgugala can you advise if this archive is created with every build, or if it was a one-off thing?

tcal-x, 2020-06-30:
I figured how to find recent install tars, for example:
```
https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/25/20200629-082552/symbiflow-arch-defs-install-1daf645a.tar.xz
```
However, this tar does not include install/share/prjxray directories, and if I use this tar in place of the April 16 tar, then when I try to follow the instructions for building the examples, I see the following error:
```
ERROR: XRAY_DATABASE_DIR variable is unset.
```

tcal-x, 2020-06-30:
@litghost @acomodi , this is where I'm tracking 100T support in symbiflow-examples.

@rw1nkler , also, in case you have any relevant information, please let me know.

litghost, 2020-09-24:
I believe this issue can be closed, @tcal-x ?

tcal-x, 2020-09-25:
Yep, two examples, one big one small!

closed 2020-09-25 by tcal-x