chipsalliance/f4pga-examples #28: Rethink download/install method

tcal-x, 2020-07-06:
Right now we use a combination of downloading a tar.gz file, plus installing Conda packages.   There is executable code/scripts in both parts.   Is there a better way?  It seems updating the tools to include recent additions is difficult (for example, see issue #27).

Perhaps there is some reworking already in the pipeline.

tcal-x, 2020-07-07:
Alessandro is working on this I think.

WhiteNinjaZ, 2021-08-20:
@acomodi @kgugala @nelsobe So do upstream changes to the toolchain have to be manually ported into the toolchain? I noticed that the changes in symbiflow-yosys-plugins and symbiflow-arch-defs don't seem to be reflected in my toolcahin even after I reinstalled the most up to date version of symbiflow-examples.

bl0x, 2021-11-24:
It would be nice to have a one-page guide on how to update symbiflow-examples and the associated toolchain to pull in the latest changes. Right now the most straightforward way for me would be to purge and reinstall.

What are the recommendations?

umarcor, 2021-12-14:
@WhiteNinjaZ I think that yosys plugins are pinned in the `environment.yml` files:

- https://github.com/SymbiFlow/symbiflow-examples/blob/master/xc7/environment.yml#L11
- https://github.com/SymbiFlow/symbiflow-examples/blob/master/eos-s3/environment.yml#L7

And the arch-defs are pinned in the docs (see #223).