chipsalliance/f4pga-examples #37: Arty A7 100T genfasm crash (in counter_test)

Skydive, 2020-07-27:
I was told on IRC that setting VPR_SEED may be a workaround.
I've repeated the command, after deleting the build folder each time:
`TARGET="arty_100" make | tee <LOGFILE>`
1) `export VPR_SEED="--seed 1341221332"`
2) `export VPR_SEED="--seed 12345"`
3) `unset VPR_SEED`
At the end of each log:
```
Begin loading /opt/symbiflow/xc7/install/bin/vpr_common: line 111: 26523 Killed                  genfasm ${ARCH_DEF} ${EBLIF} --device ${DEVICE_NAME} ${VPR_OPTIONS} --read_rr_graph ${RR_GRAPH} $@
make: *** [Makefile:40: build/top.fasm] Error 137
```
I hope this aids efforts to fix this problem. 
[log-seed-unset.log](https://github.com/SymbiFlow/symbiflow-examples/files/4984781/log-seed-unset.log)
[log-seed-1341221332.log](https://github.com/SymbiFlow/symbiflow-examples/files/4984782/log-seed-1341221332.log)
[log-seed-12345.log](https://github.com/SymbiFlow/symbiflow-examples/files/4984783/log-seed-12345.log)

litghost, 2020-07-27:
Sorry I was mistaken about the error you have.  The error you are seeing is in genfasm, not in placement.  I'll take a look this week.

litghost, 2020-07-28:
Can you  please post the ".net", ".place" and ".route" files for you target?  I believe I know the issue, and want to replicate what you are seeing.

Skydive, 2020-07-28:
> Can you please post the ".net", ".place" and ".route" files for you target? I believe I know the issue, and want to replicate what you are seeing.

This is for the unset VPR_SEED
[top.place](https://gist.github.com/Skydive/93d71f609691434e2ec2ba464eb4a9eb#file-top-place)
[top.route](https://gist.github.com/Skydive/93d71f609691434e2ec2ba464eb4a9eb#file-top-route)
[top.net](https://gist.github.com/Skydive/93d71f609691434e2ec2ba464eb4a9eb#file-top-net)

litghost, 2020-07-28:
So I've identified the problem.  The genfasm step for the 100T currently consumes ~7 GB of RAM, and I believe you mentioned that you had ~7 GB of RAM total.  Lowering the peak memory usage is something we are working on, but it isn't something we have a concrete path forward on.  Sorry for the inconvenience.

Data:
```
Load Routing took 0.20 seconds (max_rss 7172.8 MiB, delta_rss +0.0 MiB)

Checking to ensure routing is legal...
Completed routing consistency check successfully.

Serial number (magic cookie) for the routing is: 230918774
Circuit successfully routed with a channel width factor of 500.
Writing Implementation FASM: top.fasm
The entire flow of VPR took 44.8705 seconds.
FASM extra: top_fasm_extra.fasm
	Command being timed: "make build/top.fasm"
	User time (seconds): 105.96
	System time (seconds): 9.97
	Percent of CPU this job got: 103%
	Elapsed (wall clock) time (h:mm:ss or m:ss): 1:52.05
	Average shared text size (kbytes): 0
	Average unshared data size (kbytes): 0
	Average stack size (kbytes): 0
	Average total size (kbytes): 0
	Maximum resident set size (kbytes): 7344968
	Average resident set size (kbytes): 0
	Major (requiring I/O) page faults: 0
	Minor (reclaiming a frame) page faults: 5003423
	Voluntary context switches: 10117
	Involuntary context switches: 69080
	Swaps: 0
	File system inputs: 0
	File system outputs: 37128
	Socket messages sent: 0
	Socket messages received: 0
	Signals delivered: 0
	Page size (bytes): 4096
	Exit status: 0
```