chipsalliance/f4pga-examples #43: Explore converting the Travis CI to instead use GitHub actions

mithro, 2020-08-19:
@FFY00 has set up GitHub actions to generate Arch Linux packages for SymbiFlow at https://github.com/FFY00/symbiflow-arch-pkgs

This has a couple of really nice aspects;
 - It generates a list of tasks to run based on which packages have been modified.
 - It can potentially have a much clearer dependency tree described.

It seems that GitHub actions are the future here. @FFY00 has said he would be interesting to explore using it instead.

mgielda, 2020-11-22:
As per the README note, GitHub has too little storage to fit all of SF arch defs for series 7. So until that is resolved, we are somewhat stuck with Travis.

mgielda, 2020-11-22:
Having said that I am on arch so lemme at least get this running locally ;)

kgugala, 2020-11-26:
We've split the arch definitions into separate packages for every supported device type. Also, we've split the tests into separate jobs for every device-test combination.

With the above changes we were able to use gh-actions.

closed 2020-11-26 by kgugala