chipsalliance/f4pga-examples #45: Errors while installing Toolchain for Artix-7 Devices

SSandeep19, 2020-09-04:
Hi i am new to Linux and working on to generate bitstream file for a Verilog file using Symbiflow for a Artix-7 FPGA. I successfully installed Conda in tool chain installation and moved to second part of tool chain installation for Artix-7 and i got errors as shown in the screenshot.
[image]

tcal-x, 2020-09-04:
It looks like there was a problem with the `wget` command that is the first step.   It looks like you have captured stdout in your `conda_installer.sh` file, not the contents itself.   If you look at the file, you _should_ see this:
```
$ head conda_installer.sh 
#!/bin/sh
#
# NAME:  Miniconda3
# VER:   py38_4.8.3
# PLAT:  linux-64
# LINES: 570
# MD5:   728a5c953844917b24246080968d03b8

export OLD_LD_LIBRARY_PATH=$LD_LIBRARY_PATH
unset LD_LIBRARY_PATH
```

Try rerunning the `wget`, making sure that your cut-and-paste doesn't change characters.

SSandeep19, 2020-09-04:
yes you are right and now i can see the file as you mentioned but there is still an error which is different from the previous errors.
[image]

tcal-x, 2020-09-04:
Yes, that is a mistake in the instructions.   It should be:
```
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/presubmit/install/477/20200714-082108/symbiflow-arch-defs-install-8eb88e76.tar.xz | sudo tar -xJ --one-top-level=$INSTALL_DIR/install
```
The difference is the `sudo`, which gives you root permission for the duration of the following command.

This assumes that you have `sudo` permission on your machine.  You may be asked for your password or `root`'s password.

SSandeep19, 2020-09-04:
giving sudo worked . I believe you are saying to use the wget command you have  given immediately after the sudo command because if i follow the instructions it is throwing errors. can you please confirm.
[image]

tcal-x, 2020-09-04:
I'm sorry I added sudo to the wrong instruction (which also needs sudo).  Let me edit the previous comment so that I don't confuse other readers.

This 'file not found' is another problem with the README.   It asks you to use a file that isn't there you.  You don't have that file until you do the `git clone`.

So please go ahead with the `git clone` part of the instructions:
```
git clone https://github.com/SymbiFlow/symbiflow-examples && cd symbiflow-examples
```
And then continue with 
```
conda env create -f examples/xc7/environment.yml
<etc>
```
Of course, skip the `git clone` when you get to that part of the instructions, since you've already done it.

tcal-x, 2020-09-04:
@SSandeep19 , I've tried to fix the README problems here:

[https://github.com/tcal-x/symbiflow-examples/tree/doc](https://github.com/tcal-x/symbiflow-examples/tree/doc)

After I double-check my changes, I'll submit a PR to this repo.

SSandeep19, 2020-09-07:
Hi tcal-x,

i am facing an error while executing a command (pushd litex/litex/boards/targets && ./arty.py --toolchain symbiflow --cpu-type vexriscv --build && popd) almost at the end, attaching the screenshot and i don't know if it changes something. I also would like to know few things

1. What boards of Artix-7 are supported i have Digilent NEXYS A7 will it be supported or should i change some architecture files.
2. If it supports will i be able to generate bitstream with my own constrains file and is there any documentation i can refer to 
3. It will be helpful for me to know how to run Yosys,VPR and FPGA assembler individually and entire toolchain with my own eblif files or FASAM files, is this possible with the current setup 
[image]
? and will be greatly helpful if you can point me in the right direction

tcal-x, 2020-09-08:
- The `migen` error is some installation/path issue.   I haven't tried the LiteX build in this specific example, but I've seen that error other times with LiteX.   @acomodi , @kgugala , any ideas?
- Yes, you should be able to use SymbiFlow with Nexys A7, although maybe not all examples (specifically, the Linux example, since it requires a large amount of memory, and I haven't checked the specs for Nexys A7).   You'd need to alter the pin constraints, since different boards will differ how they connect the device pins to LEDs, switches, etc. The syntax for PCF files is straightforward (for example, https://github.com/SymbiFlow/symbiflow-examples/blob/master/examples/xc7/counter_test/arty.pcf).   Digilent has master XDC files for [different boards](https://github.com/Digilent/digilent-xdc/blob/master), and you can use that information to create your own PCF file.
- The `Makefile` in each example directory does a good job breaking down the different steps.   If you are not familiar with `make`, it is worth learning the basics.

tcal-x, 2020-09-08:
P.S. you don't need to put ALL the pins in the master XDC file in your PCF file -- just the ones the design uses.

litghost, 2020-09-08:
In general, `sudo` shouldn't be required for using anything from https://github.com/SymbiFlow/symbiflow-examples, as everything should be installed into user local folders.  Which commands/steps are believed to require `sudo`?

litghost, 2020-09-08:
If I had to guess, the fact that:
```
INSTALL_DIR="/opt/symbiflow/xc7"
```

is part of `/opt` is the issue.  Rather than rooting the install to `/opt` it probably just belongs in the users home folder by default, like:

```
INSTALL_DIR="~/symbiflow/xc7"
```

and then the `sudo` stuff can be dropped.  If the user wants to install into a folder off of root, then can handle the permissions independent of the instructions.

litghost, 2020-09-24:
@SSandeep19 The instructions have been updated such that the sudo shouldn't be required!  We believe this issue is fixed, so I'm going to close this issue.  If you find you still have an issue, let us know!

closed 2020-09-24 by litghost