chipsalliance/f4pga-examples #48: Travis is always downloading master revision when executing tests.

kamilrakoczy, 2020-09-08:
Travis is using tuttest to test tutorial commands in the README.md.

Tuttest is following commands from the README.md and make sure they are working.
In the README.md there is a command that is cloning repository and entering newly cloned repository:
```
git clone https://github.com/SymbiFlow/symbiflow-examples && cd symbiflow-examples
```
this means, that when someone opens PR, changes from this PR are not reflected in the tests. Always master version of the repository is used in tests.

litghost, 2020-09-18:
@tcal-x encountered this too, my suggestion to fix it is https://github.com/SymbiFlow/symbiflow-examples/pull/50#issuecomment-695023303

tcal-x, 2020-09-21:
@kamilrakoczy -- if you have not started on the fixes suggested in Keith's comment, or plan to start soon, I'll take a whack at it.

kamilrakoczy, 2020-09-22:
@tcal-x No, I didn't started this, feel free to fix this.

litghost, 2020-09-24:
This was fixed in #50

closed 2020-09-24 by litghost