chipsalliance/f4pga-examples #54: How to install from sources and where is the doc about the commands?

rodrigomelo9, 2020-09-29:
Hello. I want to know from which repository are the symbiflow tools (symbiflow_synth, symbiflow_pack, etc) generated. Also, I want their documentation. Is enough to point me to the right doc. Are Yosys+nextpnr+icestrom tools, for example, completely replaced by the symbiflow tools? (in this repository, there are only examples of Xilinx and QuickLogic).

Thanks,
Rodrigo

mithro, 2020-09-29:
The packages in conda are generated using the repo at https://github.com/SymbiFlow/conda-packages

mithro, 2020-09-29:
@kgugala Could you help @rodrigomelo9 to find these?

mithro, 2020-09-29:
@rw1nkler - I think this means we need some documentation updates?

rodrigomelo9, 2020-09-30:
Hi @mithro, thanks. In the meantime, are the commands symbiflow_synth, symbiflow_pack, etc, exactly the same as yosys, nextpnr (-ice40 or -ecp5, according to the case), etc? Or are they new binaries with any difference?

> @rw1nkler - I think this means we need some documentation updates?

I think that this info is missing, and must be put somewhere at https://symbiflow.readthedocs.io/en/latest/toolchain-desc.html
If I undestand, I can help :-D I want to add symbiflow to [PyFPGA](https://gitlab.com/rodrigomelo9/pyfpga) and add an small tutorial into [FOSS for FPGA](https://github.com/rodrigomelo9/FOSS-for-digital-HW-design).

mithro, 2021-04-12:
The scripts can be found at https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc/xc7/toolchain_wrappers and usage examples at http://symbiflow-examples.rtfd.io/

rodrigomelo9, 2021-04-12:
It solves where to get the tools and doc. Thanks!

closed 2021-04-12 by mithro