chipsalliance/f4pga-examples #58: Error with "make README.rst":  "TypeError: Got secondary option for non boolean flag."

tcal-x, 2020-10-13:
This only affects you if you're updating README.rst.

To reproduce, remove README.rst (`rm README.rst`), then run `make README.rst`.   After packages are downloaded and the conda env is entered, I see this:

```
touch /home/tim/symbiflow/symbiflow-examples/env/conda/envs/symbiflow-examples/bin/python
source /home/tim/symbiflow/symbiflow-examples/env/conda/bin/activate && conda activate symbiflow-examples && rst_include include README.src.rst - > README.rst
Traceback (most recent call last):
  File "/home/tim/symbiflow/symbiflow-examples/env/conda/envs/symbiflow-examples/bin/rst_include", line 5, in <module>
    from rst_include.rst_include_cli import cli_main
  File "/home/tim/symbiflow/symbiflow-examples/env/conda/envs/symbiflow-examples/lib/python3.8/site-packages/rst_include/rst_include_cli.py", line 38, in <module>
    def cli_main(traceback: Optional[bool] = None) -> None:
  File "/home/tim/.local/lib/python3.8/site-packages/click/decorators.py", line 170, in decorator
    _param_memo(f, OptionClass(param_decls, **attrs))
  File "/home/tim/.local/lib/python3.8/site-packages/click/core.py", line 1511, in __init__
    raise TypeError('Got secondary option for non boolean flag.')
TypeError: Got secondary option for non boolean flag.
make: *** [Makefile:5: README.rst] Error 1
```

I'm running Ubuntu 20.04.1 LTS.

tcal-x, 2020-10-13:
After cleaning out symbiflow-examples, then manually going through the toolchain installation steps again, and activating the conda 'xc7' environment, I was able to `make README.rst` without seeing the same error.

So, I'll close this.   However, the `make README.rst` flow seems strange.   Even when I just installed and entered a brand new conda environment, doing `make README.rst` redoes the entire conda environment again.

closed 2020-10-13 by tcal-x

mithro, 2020-10-13:
@tcal-x - Can you log a bug to investigate that?