chipsalliance/f4pga-examples #63: Add Nexys Video examples into the repository

mithro, 2020-11-16:


mithro, 2020-11-16:
Blinky is already in fpga-perf-tool and symbiflow-arch-defs?

mithro, 2020-11-16:
https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc/xc7/tests/soc/ibex

andrewb1999, 2020-11-17:
@mithro Is there a tar available with a stable a200t rr graph/architecture?  I'm interested in getting some other people to use the nexys video with symbiflow, so I would be willing to add some examples here when I have time.

mithro, 2020-11-17:
@andrewb1999 - I'm wrong person to ask. Maybe @acomodi / @mkurc-ant / @rw1nkler can answer?

acomodi, 2020-11-23:
@andrewb1999 The latest tars produced by symbiflow-arch-defs master do contain the 200T data files ([latest](https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/98/20201120-093358/symbiflow-arch-defs-install-d5f8ce8d.tar.xz))

tcal-x, 2021-01-29:
I've built the counter example for the Nexys Video board, but now I'm not sure how to proceed.

On https://symbiflow-examples.readthedocs.io/en/latest/building-examples.html -- does the same OpenOCD config file work with the Nexys Video board?   How should JP3 be configured, "SD" or "USB"?  What about JP4, "QSPI", "JTAG", "USB/SD"?

We also need additions to the [Running example designs](https://symbiflow-examples.readthedocs.io/en/latest/running-examples.html) page for the Nexys Video board -- it's more confusing since there many more connectors.   I guess the connecting via UART can wait until the Ibex/LiteX examples are ready.