chipsalliance/f4pga-examples #66: xc7-litex failing

scted, 2020-11-22:
pre-req is fixissue 65 to get tests running

progress so far fixing is that the litex stuff is being installed outside of the symbiflow-examples directory so a cd needs to happen ... see line 5
[image]

i would prefer if the install instructions gave the user a measure control on where all the Litex stuff ends up ... needs a $LITEX_INSTALL_DIR

with the cd fix, the test is now failing in litex/litex/build/xilinx/symbiflow.py", line 103, in     ####_run_make
```python3 -m litex.soc.software.memusage bios.elf /home/travis/build/scted/litex/litex/boards/targets/build/arty/software/bios/../include/generated/regions.ld riscv64-unknown-elf
ROM usage: 20.96KiB 	(65.50%)
RAM usage: 1.63KiB 	(20.41%)
make: Leaving directory '/home/travis/build/scted/litex/litex/boards/targets/build/arty/software/bios'
Traceback (most recent call last):
  File "./arty.py", line 162, in <module>
    main()
  File "./arty.py", line 155, in main
    builder.build(**builder_kwargs, run=args.build)
  File "/home/travis/build/scted/litex/litex/soc/integration/builder.py", line 217, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/travis/build/scted/litex/litex/soc/integration/soc.py", line 1054, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/travis/build/scted/litex/litex/build/xilinx/platform.py", line 53, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/travis/build/scted/litex/litex/build/xilinx/symbiflow.py", line 263, in build
    _run_make()
  File "/home/travis/build/scted/litex/litex/build/xilinx/symbiflow.py", line 103, in _run_make
    if tools.subprocess_call_filtered(shell + [script], common.colors) != 0:
NameError: name 'shell' is not defined`

pgielda, 2020-11-22:
Indeed this seems to fail on ubuntu:trusty and old centos but not on other OSes? (other OSes have travis timeouts but that is much later in the run)

scted, 2020-11-22:
i just tested it on Ubuntu:focal

scted, 2020-11-23:
still not passing ubuntu:focal fwiw

erroring out for the directory issue:
[image]

i assume it will fail later in symbiflow.py line 103 (see above for Traceback) which is

`if tools.subprocess_call_filtered(shell + [script], common.colors) != 0:`

i can't find reference to 'shell', 'script' or 'common' anywhere in symbiflow.py

acomodi, 2020-11-23:
>  assume it will fail later in symbiflow.py line 103 (see above for Traceback) which is

if tools.subprocess_call_filtered(shell + [script], common.colors) != 0:

i can't find reference to 'shell', 'script' or 'common' anywhere in symbiflow.py

Hi @scted, this issue has been solved in litex upstream, and the error should not appear anymore if you get the latest litex version (I think you can do .`/litex_setup.py update install` to get the latest)

scted, 2020-11-23:
@acomodi 

i still can't get it to complete in Travis (see bottom for Traceback) ... now failing line 84 ... apologies if it is usererror on my part

[image]

tools.subprocess.py not returning 0 ... way beyond my capability to troubleshoot

[image]

[image]
 
still had to modify the xc7/README.rst to cd up one level to get this to get this far

rw1nkler, 2020-11-30:
@scted I think that we resolved all the problems here (according to https://github.com/SymbiFlow/symbiflow-examples/pull/73#issuecomment-734935737) and we can close this issue. Is that true?

closed 2020-11-30 by scted

scted, 2020-11-30:
@rw1nkler FYI, there is a discrepancy between the Litex arty platform and the new migen arty_a7 platform ... was using that because i am using migen to build my own stuff ... the migen stuff almost works but it says the device is 'xc7a35tic324-1L' ... symbiflow was failing at step 8 of synth ... i changed the device to 'xc7a35tic324-1' and everything worked ... not sure where to report that ... lot of new migen stuff

rw1nkler, 2020-11-30:
@scted Thank you for reporting this! This is a good place to report any problem with the SymbiFlow Toolchain.
I would suggest opening a separate issue for that, to track the problem. (This issue is closed)
In fact, I probably know what is causing the problem. Can you please open a separate issue with the information that you gave above? Can you also provide more detailed information on how to reproduce your problem?