chipsalliance/f4pga-examples #82: Migen - Cat() simulation not matching verilog when Cat object is sliced

scted, 2020-12-06:
Have spent a week on this so far.  Cat() not behaving as i expect when i use slices Cat_object[slice] ... 

Have tried a number of permutations ... assigning to a slice  ... Cat_object[n].eq(rhs) or from a slice ... lhs.eq(Cat_object[n])

The verilog (good and bad) has been tested in HW with both Vivado and Symbiflow ... it is not the toolchain that is a problem.

Possible that I do not understand how Cat is supposed to be used but the fact remains the simulation results differ from the produced verilog.

Code I used to test here:  https://github.com/scted/Cat-demo

In the code snippet (bottom), the 'lhs' and 'rhs' permutations (where slicing of the Cat_object is used) do not connect all the ins to the outs ... only the msb ends up being connect

The verilog producted for 'lhs' and 'rhs' (below) does something funky with 'slice_proxy0' and 'slice_proxy1' ...

```
Writing verilog for "rhs" ... something gets mangled and only the MSB is connected

/* Machine-generated using Migen */
module top(

);

reg [1:0] ins = 2'd0;
reg [1:0] outs;
wire s0;
wire s1;
wire [1:0] slice_proxy0;
wire [1:0] slice_proxy1;

// synthesis translate_off
reg dummy_s;
initial dummy_s <= 1'd0;
// synthesis translate_on

assign {s1, s0} = ins;

// synthesis translate_off
reg dummy_d;
// synthesis translate_on
always @(*) begin
	outs <= 2'd0;
	outs[0] <= slice_proxy0[0];
	outs[1] <= slice_proxy1[1];
// synthesis translate_off
	dummy_d <= dummy_s;
// synthesis translate_on
end
assign slice_proxy0 = {s1, s0};
assign slice_proxy1 = {s1, s0};

endmodule
``` 

```    

        '''
        Different ways to implement ins(2) => cat_bus(=Cat(s0, s1)) => outs(2)
        sim the same in all cases: ins == outs
        
        "no_slicing" :: verilog == sim
        "lhs" : assign the lhs of bus from ins via slicing: verilog <> sim
        "rhs" : assign the rhs of bus to outs via slicing: verilog <> sim
        "use_signals" : assign ins/out directly to/from cat_bus: verilog == sim
        '''
        
        
        self.ins = ins = Signal(2)
        self.outs = outs = Signal(2)

        self.s0 = s0 = Signal()                                 
        self.s1 = s1 = Signal()
        
        #object of interest
        self.cat_bus = cat_bus = Cat(s0, s1)

        if permutation == "no_slicing":         #sim == verilog
            self.comb += cat_bus.eq(ins)
            
            self.comb += outs.eq(cat_bus)

        elif permutation == "lhs":              #sim <> verilog
            self.comb += cat_bus[0].eq(ins[0])
            self.comb += cat_bus[1].eq(ins[1])
            
            self.comb += outs.eq(cat_bus)

        elif permutation == "rhs":              #sim <> verilog
            self.comb += cat_bus.eq(ins)

            self.comb += outs[0].eq(cat_bus[0])
            self.comb += outs[1].eq(cat_bus[1])

        elif permutation == "use_signals":      #sim == verilog
            self.comb += s0.eq(ins[0])
            self.comb += s1.eq(ins[1])
            
            self.comb += outs[0].eq(s0)
            self.comb += outs[1].eq(s1)

```

mithro, 2020-12-06:
Hi @scted,

This sounds like a Migen issue, not something that is specific to symbiflow-examples? Migen issues can be reported as https://github.com/m-labs/migen

Thanks,

Tim @mithro Ansell

scted, 2020-12-06:
almost certainly migen ... was told to report anything remotely related to the examples (using the migen installed by the examples) here but will also post on m-labs

scted, 2020-12-08:
@mithro FYI ... https://github.com/m-labs/migen/issues/228

real issue with a workaround ... which is to assign the Cat object to an intermediate signal of equal size and then slice the intermediate signal

do you know anything about ghdl? ...  "Is this any good? https://github.com/ghdl/ghdl-yosys-plugin"

m-labs think emitting vhdl would be better ... seems like verilog may be difficult and errorprone

do i close this issue here or do you?  #githubrookie

kgugala, 2020-12-08:
Hi @scted I did some a fast check how your example behaves in [nMigen](https://github.com/nmigen/nmigen). Had to tweak the code a little to nMigen compliant:

```
from nmigen import *
from nmigen.back import verilog

class TestRHSCat(Elaboratable):

    def __init__(self):
        self.ins = Signal(2)
        self.outs = Signal(2)

        self.s0 = Signal()
        self.s1 = Signal()

    def elaborate(self, platform):
        m = Module()
        #Cat_object of interest
        cat_bus = Cat(self.s0, self.s1)
        m.d.comb += cat_bus.eq(self.ins)

        #using using slices of cat_bus
        m.d.comb += self.outs[0].eq(cat_bus[0])
        m.d.comb += self.outs[1].eq(cat_bus[1])

        return m

if __name__ == "__main__":
    rhscat = TestRHSCat()
    print(verilog.convert(rhscat, ports=[rhscat.ins, rhscat.outs]))
```

And I got the following Verilog:

```
/* Generated by Yosys 0.9+3667 (git sha1 e7f36d01, clang 10.0.0 -fPIC -Os) */

(* \nmigen.hierarchy  = "top" *)
(* top =  1  *)
(* generator = "nMigen" *)
module top(outs, ins);
  (* src = "ntest.py:7" *)
  input [1:0] ins;
  (* src = "ntest.py:8" *)
  output [1:0] outs;
  (* src = "ntest.py:10" *)
  wire s0;
  (* src = "ntest.py:11" *)
  wire s1;
  assign outs[1] = s1;
  assign outs[0] = s0;
  assign { s1, s0 } = ins;
endmodule
```
The generated Verilog is much simpler than the one from migen. Also it looks correct - slicing worked fine here.

scted, 2020-12-08:
@kgugala  ... cool  ... might give nMigen a try and see what i can break:)  ... interesting the Yoysy is generating the code.

closed 2020-12-08 by scted