chipsalliance/f4pga-examples #106: issues with xc7 instructions for openocd

tcal-x, 2020-12-18:
If this addressed by a PR in flight, just close this.

I'm going through the examples at https://symbiflow-examples.readthedocs.io/en/latest/building-examples.html .

After the counter example is built, the user is told to execute the following to program the part with the bitstream:

```
openocd -f ${INSTALL_DIR}/conda/share/openocd/scripts/board/digilent_arty.cfg -c "init; pld load 0 top.bit; exit"
```

There are two things wrong with this:
* `digilent_arty.cfg` is not at the provided path.  In fact I cannot find it at all in the installed tool directory.  I have only installed the xc7 tools.
* the user is still in the xc7/ directory, so either they need to `cd` to the correct build directory, or `top.bit` should be replaced with the path to the correct file

kgugala, 2020-12-18:
can we rewrite the guide to use https://github.com/trabucayre/openFPGALoader? This tool is way simpler to use than openOCD and it supports xc7 boards we target

tcal-x, 2020-12-18:
I'd be willing to give it a try if it simplifies things overall.
* is it easy to add to the downloads/packages for the user, or can they install it with apt-get?
* it has an AGPL license -- is that an issue?

kgugala, 2020-12-18:
> it has an AGPL license -- is that an issue?

this is not an issue if we just use it as a third party tool. (BTW openocd is GPL licensed)

mithro, 2020-12-18:
@kgugala - I'm afraid we can't touch anything which is **A**GPL. GPL is fine.

mithro, 2020-12-18:
https://opensource.google/docs/using/agpl-policy/

kgugala, 2020-12-18:
oh, OK. So we need to fix openocd instructions then

mithro, 2020-12-18:
@kgugala - If anyone has a good relationship with the openFPGALoader people, it might be worth talking to them and seeing if they would consider "downgrading" to just GPL (rather than AGPL) -- which means we could then use them. I'm not sure the AGPL makes a lot of sense for their software anyway - the additions in AGPL are mainly focused towards "web" technologies like databases, web apps, etc.

pgielda, 2020-12-18:
Seems its more than one person but the main committer is gwenhael.goavec-merou@trabucayre.com

pabigot, 2020-12-19:
In case it's relevant:  `digilent_arty.cfg` seems to be found in the openocd that comes from the litex::openocd conda package, which is incompatible with python 3.8 which is installed following the directions in this repository.

I'm not seeing any obvious difference between `digilent_arty.cfg` and the `arty_s7.cfg` in upstream OpenOCD which can be installed outside of conda.  Although I've successfully programmed my Arty A7-100T with `arty_s7.cfg` in the past, and substituting it in the example command (after first `cd counter_test/build/arty_100/`) produces no errors, I don't see any flashing LEDs.

acomodi, 2020-12-21:
@pabigot @tcal-x I have opened a [PR](https://github.com/SymbiFlow/symbiflow-examples/pull/110) to:

- fix the absence of openocd in the env (and also the link to the config file);
- fix the non-working counter test for arty. There was a typo in the constraints files which messed up the IOSTANDARD values of the LEDs IOPADs.