chipsalliance/f4pga-examples #108: Error encountered with FASM, antlr_to_tuple, running xc7 counter example

tcal-x, 2020-12-20:
After a clean install of xc7 tools according to the symbiflow-examples instructions, I get to the first example to run (https://symbiflow-examples.readthedocs.io/en/latest/building-examples.html):
```
TARGET="arty_35" make -C counter_test
```
and this runs Symbiflow but ends with this error:
```
......
Serial number (magic cookie) for the routing is: 184139735
Circuit successfully routed with a channel width factor of 500.
Incr Slack updates 1 in 1.4195e-05 sec
Full Max Req/Worst Slack updates 1 in 3.823e-06 sec
Incr Max Req/Worst Slack updates 0 in 0 sec
Incr Criticality updates 0 in 0 sec
Full Criticality updates 1 in 1.8115e-05 sec
Writing Implementation FASM: top.fasm
The entire flow of VPR took 19.0291 seconds.
FASM extra: top_fasm_extra.fasm
writing final fasm
cd build/arty_35 && symbiflow_write_bitstream -d artix7 -f top.fasm -p xc7a35tcsg324-1 -b top.bit
Writing bitstream ...
/media/tim/scratch/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/fasm/parser/__init__.py:27: RuntimeWarning: 
Falling back on slower textX parser implementation:
  ImportError: cannot import name 'antlr_to_tuple' from 'fasm.parser' (/media/tim/scratch/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/fasm/parser/__init__.py)
Please install all dependencies and reinstall with:
  pip uninstall
  pip install -v fasm
  '  pip install -v fasm'.format(e), RuntimeWarning)
make: Leaving directory '/home/tim/symbiflow/symbiflow-examples/xc7/counter_test'
```
Maybe things are just temporarily out of sync.   Feel free to close this if this is a known/expected issue.
FYI @HackerFoo in case you have any information.

In case it matters, I'm on Ubuntu 20.04.

HackerFoo, 2020-12-20:
Could you please try:

```
pip uninstall fasm
pip install -v git+https://github.com/SymbiFlow/fasm.git
```

 and post the results? Thanks.

tcal-x, 2020-12-20:
Result:
```
$ pip uninstall fasm
Found existing installation: fasm 0.0.2
Uninstalling fasm-0.0.2:
  Would remove:
    /media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/bin/fasm
    /media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/fasm-0.0.2.dist-info/*
    /media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/fasm/*
Proceed (y/n)? y
  Successfully uninstalled fasm-0.0.2

$ pip install -v git+https://github.com/SymbiFlow/fasm.git
<missed lots of output>
  creating '/tmp/pip-wheel-6ovty67f/tmph7ojlh5x/fasm-0.0.2-cp37-cp37m-linux_x86_64.whl' and adding 'build/bdist.linux-x86_64/wheel' to it
  adding 'fasm/__init__.py'
  adding 'fasm/model.py'
  adding 'fasm/output.py'
  adding 'fasm/tool.py'
  adding 'fasm/parser/__init__.py'
  adding 'fasm/parser/antlr.py'
  adding 'fasm/parser/antlr_to_tuple.pyx'
  adding 'fasm/parser/fasm.tx'
  adding 'fasm/parser/textx.py'
  adding 'fasm-0.0.2.dist-info/METADATA'
  adding 'fasm-0.0.2.dist-info/WHEEL'
  adding 'fasm-0.0.2.dist-info/entry_points.txt'
  adding 'fasm-0.0.2.dist-info/top_level.txt'
  adding 'fasm-0.0.2.dist-info/RECORD'
  removing build/bdist.linux-x86_64/wheel
  Traceback (most recent call last):
    File "setup.py", line 82, in run
      super().run()
    File "/tmp/pip-build-env-kd1l85w3/overlay/lib/python3.7/site-packages/setuptools/command/build_ext.py", line 79, in run
      _build_ext.run(self)
    File "/tmp/pip-build-env-kd1l85w3/overlay/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
      _build_ext.build_ext.run(self)
    File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/distutils/command/build_ext.py", line 340, in run
      self.build_extensions()
    File "/tmp/pip-build-env-kd1l85w3/overlay/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
      _build_ext.build_ext.build_extensions(self)
    File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
      self._build_extensions_serial()
    File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
      self.build_extension(ext)
    File "setup.py", line 139, in build_extension
      env=env)
    File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/subprocess.py", line 363, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['cmake', '/tmp/pip-req-build-snybwo18/src', '-DCMAKE_INSTALL_PREFIX=/tmp/pip-req-build-snybwo18/build/lib.linux-x86_64-3.7/fasm/parser', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-req-build-snybwo18/build/lib.linux-x86_64-3.7/fasm/parser', '-DPYTHON_EXECUTABLE=/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/bin/python3.7', '-DANTLR_RUNTIME_TYPE=static', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
  Building wheel for fasm (PEP 517) ... done
  Created wheel for fasm: filename=fasm-0.0.2-cp37-cp37m-linux_x86_64.whl size=15106 sha256=fa6b3a17ecca8b70538e7599e5f0d37e5e40d747ad4b2b65cd5d8552c749eaec
  Stored in directory: /tmp/pip-ephem-wheel-cache-g8fluwhr/wheels/32/eb/40/468bb90814511ea94a44dc6f8c355c8936e02d58d24eb59e9c
Successfully built fasm
Installing collected packages: fasm
  7124d4f2239065886dbcf80e5e3ba289a8fec1c1

  changing mode of /media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/bin/fasm to 775
Successfully installed fasm-0.0.2
Removed build tracker: '/tmp/pip-req-tracker-rq3rmv_9'
```
Then
```
TARGET="arty_35" make -C counter_test
```
but the result is the same
```
The entire flow of VPR took 19.3198 seconds.
FASM extra: top_fasm_extra.fasm
writing final fasm
cd build/arty_35 && symbiflow_write_bitstream -d artix7 -f top.fasm -p xc7a35tcsg324-1 -b top.bit
Writing bitstream ...
/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/fasm/parser/__init__.py:27: RuntimeWarning: 
Falling back on slower textX parser implementation:
  ImportError: cannot import name 'antlr_to_tuple' from 'fasm.parser' (/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/fasm/parser/__init__.py)
Please install all dependencies and reinstall with:
  pip uninstall
  pip install -v fasm
  '  pip install -v fasm'.format(e), RuntimeWarning)
make: Leaving directory '/home/tim/symbiflow/symbiflow-examples/xc7/counter_test'
```

HackerFoo, 2020-12-20:
The ANTLR parser has some dependencies outside of PyPI, which is why the fallback parser is necessary.

Could you please post the results of:

```
sudo apt install cmake default-jre-headless uuid-dev
https://github.com/SymbiFlow/fasm.git
cd fasm
make build
```

Thanks.

tcal-x, 2020-12-20:
...descending into package hell...
```
$ sudo apt install cmake default-jre-headless uuid-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
default-jre-headless is already the newest version (2:1.11-72).
uuid-dev is already the newest version (2.34-0.1ubuntu9.1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 cmake : Depends: libssl1.0.0 (>= 1.0.2~beta3) but it is not installable
E: Unable to correct problems, you have held broken packages.

$ sudo apt install libssl1.0.0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libssl1.0.0 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libssl1.0.0' has no installation candidate

```
If I go ahead with the clone and attempt to build:
```
$ $ make build
if [ -e env/bin/activate ]; then . env/bin/activate; fi; python setup.py build
Traceback (most recent call last):
  File "setup.py", line 24, in <module>
    from Cython.Build import cythonize
ModuleNotFoundError: No module named 'Cython'
make: *** [Makefile:45: build] Error 1
```

andrewb1999, 2020-12-20:
I also run into the same issue with antlr_to_tuple, so it's not an isolated issue.

mithro, 2020-12-20:
Any idea why this isn't appearing in the CI?

HackerFoo, 2020-12-20:
You'll definitely need `cmake`. Once you have a working `cmake`, `pip install -v git+https://github.com/SymbiFlow/fasm.git` should install the Cython requirement automatically.

@mithro CI has `cmake` intalled: https://github.com/SymbiFlow/fasm/blob/master/.github/workflows/presubmit.yml#L62

HackerFoo, 2020-12-20:
I think the warning would be more helpful if it just referred to the [build instructions](https://github.com/symbiflow/fasm#build-instructions).

andrewb1999, 2020-12-20:
@HackerFoo I do have cmake installed.  Here is the output of `pip install -v git+https://github.com/SymbiFlow/fasm.git`:
```
  Traceback (most recent call last):
    File "setup.py", line 82, in run
      super().run()
    File "/tmp/pip-build-env-jkd3kdm5/overlay/lib/python3.7/site-packages/setuptools/command/build_ext.py", line 79, in run
      _build_ext.run(self)
    File "/tmp/pip-build-env-jkd3kdm5/overlay/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
      _build_ext.build_ext.run(self)
    File "/tools/symbiflow/xc7/conda/envs/xc7/lib/python3.7/distutils/command/build_ext.py", line 340, in run
      self.build_extensions()
    File "/tmp/pip-build-env-jkd3kdm5/overlay/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
      _build_ext.build_ext.build_extensions(self)
    File "/tools/symbiflow/xc7/conda/envs/xc7/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
      self._build_extensions_serial()
    File "/tools/symbiflow/xc7/conda/envs/xc7/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
      self.build_extension(ext)
    File "setup.py", line 139, in build_extension
      env=env)
    File "/tools/symbiflow/xc7/conda/envs/xc7/lib/python3.7/subprocess.py", line 363, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['cmake', '/tmp/pip-req-build-xe_txdcp/src', '-DCMAKE_INSTALL_PREFIX=/tmp/pip-req-build-xe_txdcp/build/lib.linux-x86_64-3.7/fasm/parser', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-req-build-xe_txdcp/build/lib.linux-x86_64-3.7/fasm/parser', '-DPYTHON_EXECUTABLE=/tools/symbiflow/xc7/conda/envs/xc7/bin/python', '-DANTLR_RUNTIME_TYPE=static', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
  Building wheel for fasm (PEP 517) ... done
  Created wheel for fasm: filename=fasm-0.0.2-cp37-cp37m-linux_x86_64.whl size=15106 sha256=8b8fae0bd252a960a71da20003d14952492f0cd3a2dbe9eda2f4ddce28ad29bf
  Stored in directory: /tmp/pip-ephem-wheel-cache-w21x4jn0/wheels/32/eb/40/468bb90814511ea94a44dc6f8c355c8936e02d58d24eb59e9c
Successfully built fasm
Installing collected packages: fasm
  7124d4f2239065886dbcf80e5e3ba289a8fec1c1

  changing mode of /tools/symbiflow/xc7/conda/envs/xc7/bin/fasm to 755
Successfully installed fasm-0.0.2
Removed build tracker: '/tmp/pip-req-tracker-2bowmsdc'
``

tcal-x, 2020-12-20:
> You'll definitely need `cmake`. 

I do have cmake -- as a Python module.

```
$ which cmake
/home/tim/.local/bin/cmake
$ pip list | grep cmake
cmake                  3.16.3
$ pip3 list | grep cmake
cmake                  3.16.3
```

In other instructions that asked me to `sudo apt install cmake <other packages>`, I'd just skip the cmake and it would be ok.

But, let me get to the bottom of why the apt install of cmake doesn't work for me.

HackerFoo, 2020-12-20:
@andrewb1999 I can see that `cmake` failed, but unfortunately not why. This should print the output from `cmake`:
```
git clone https://github.com/SymbiFlow/fasm.git
cd fasm
make build
```

mithro, 2020-12-20:
[image]

andrewb1999, 2020-12-20:
@HackerFoo Looks like it's not getting the cython dependency correctly (same issue as @tcal-x )
```
if [ -e env/bin/activate ]; then . env/bin/activate; fi; python setup.py build
Traceback (most recent call last):
  File "setup.py", line 24, in <module>
    from Cython.Build import cythonize
ModuleNotFoundError: No module named 'Cython'
make: *** [Makefile:45: build] Error 1
```

mithro, 2020-12-20:
`python setup.py` won't install dependencies, all dependencies must be preinstalled. You probably want something like `pip install .` if you want packages to be installed.

tcal-x, 2020-12-21:
My problems installing `cmake` with `apt` seem to have originated from having "kitware" as a source.   After I removed that, I was able to install `cmake`.   I'll try the install again and see if it goes any better.

tcal-x, 2020-12-21:
I ran this to take care of the Cython requirement:
```
pip3 install cython
```
Then I followed the instructions on the page https://github.com/symbiflow/fasm#build-instructions including the `sudo apt install ... libantlr4-runtime-dev` and the `git submodule update --init`, and after that, the `make build` succeeded.   I did this install (also from the page mentioned earlier) while still in the conda environment ("xc7"):

```
pip install . --install-option="--antlr-runtime=shared" --no-use-pep517
```
and then the original SymbiFlow build of the counter example worked.

litghost, 2021-01-05:
What is the status of this?

HackerFoo, 2021-01-05:
The Conda environment.yml should provide all necessary dependencies after https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1908

HackerFoo, 2021-01-05:
It looks like the install instructions specify an older version of the toolchain. Should this be automatically updated somehow? There are "latest" text files that specify the latest URL: https://github.com/litex-hub/litex-conda-eda/blob/master/pnr/vtr-optimized/build.sh#L34-L53

tcal-x, 2021-01-19:
I've assigned this to myself to check if the issue has been resolved.

tcal-x, 2021-01-20:
Nothing has changed; I still see the original error.   

It seems we either need to address #115, or, as a band-aid, manually update the versions.

tcal-x, 2021-01-24:
Correction: the final warning from fasm looks like failure, but in fact **a bitstream is produced**, and it works correctly on the Arty 35T board (`counter_test` example).   

I had noticed in Quicklogic guides for QORC, they specifically describe this warning and tell the user that it's ok.   

This is the warning:
```
FASM extra: top_fasm_extra.fasm
writing final fasm
cd build/arty_35 && symbiflow_write_bitstream -d artix7 -f top.fasm -p xc7a35tcsg324-1 -b top.bit
Writing bitstream ...
/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/fasm/parser/__init__.py:27: RuntimeWarning: 
Falling back on slower textX parser implementation:
  ImportError: cannot import name 'antlr_to_tuple' from 'fasm.parser' (/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/fasm/parser/__init__.py)
Please install all dependencies and reinstall with:
  pip uninstall
  pip install -v fasm
  '  pip install -v fasm'.format(e), RuntimeWarning)
make: Leaving directory '/home/tim/symbiflow/symbiflow-examples/xc7/counter_test'
```

mithro, 2021-01-25:
@tcal-x - Yes, that is a warning but means the command will be much slower. We should make sure the fast path is what most people get by default.

jgoeders, 2021-02-01:
I was also running into this.  Although there is a fallback, the warning would be printed to my fasm file, which would then break things:
```
...
LIOI3_SING_X0Y100.ILOGIC_Y0.ISERDES.NUM_CE.N1
LIOI3_SING_X0Y100.OLOGIC_Y0.OMUX.D1
LIOI3_SING_X0Y100.OLOGIC_Y0.OQUSED
LIOI3_SING_X0Y100.OLOGIC_Y0.OSERDES.DATA_RATE_TQ.BUF
Falling back on slower textX parser implementation:
  ImportError: cannot import name 'antlr_to_tuple' from partially initialized module 'fasm.parser' (most likely due to a circular import) (/home/jgoeders/bfasst/third_party/fasm2bels/env/lib/python3.8/site-packages/fasm/parser/__init__.py)
Please install all dependencies and reinstall with:
  pip uninstall
  pip install -v fasm
  warn(
```

I followed @HackerFoo 's instructions and did the following:
```
git clone https://github.com/SymbiFlow/fasm.git
cd fasm
make build
```

After which I realized I needed to:

1. Make `python` point to `python3`.  (I'm using Ubuntu 20 where `python` points to nothing by default.  I just added `alias python=python3` to my `~/.bashrc`).
2. `sudo apt install python3-venv`

This fixed thing for me and the fasm.parser ImportError is gone.  Hope this helps others.

tcal-x, 2021-02-01:
in fasm/, "make build" is now giving me this:

```
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:23 (include):
  include could not find load file:

    ExternalAntlr4Cpp

CMake Error at CMakeLists.txt:47 (find_package):
  By not providing "FindANTLR.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ANTLR", but
  CMake did not find one.

  Could not find a package configuration file provided by "ANTLR" with any of
  the following names:

    ANTLRConfig.cmake
    antlr-config.cmake

  Add the installation prefix of "ANTLR" to CMAKE_PREFIX_PATH or set
  "ANTLR_DIR" to a directory containing one of the above files.  If "ANTLR"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/tim/symbiflow/symbiflow-examples/fasm/build/temp.linux-x86_64-3.7/CMakeFiles/CMakeOutput.log".
Failed to build ANTLR parser, falling back on slower textX parser. Error:
 Command '['cmake', '/home/tim/symbiflow/symbiflow-examples/fasm/src', '-DCMAKE_INSTALL_PREFIX=/home/tim/symbiflow/symbiflow-examples/fasm/build/lib.linux-x86_64-3.7/fasm/parser', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/tim/symbiflow/symbiflow-examples/fasm/build/lib.linux-x86_64-3.7/fasm/parser', '-DPYTHON_EXECUTABLE=/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/bin/python', '-DANTLR_RUNTIME_TYPE=static', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "setup.py", line 82, in run
    super().run()
  File "/home/tim/miniconda3/lib/python3.7/site-packages/setuptools/command/build_ext.py", line 79, in run
    _build_ext.run(self)
  File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "setup.py", line 139, in build_extension
    env=env)
  File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '/home/tim/symbiflow/symbiflow-examples/fasm/src', '-DCMAKE_INSTALL_PREFIX=/home/tim/symbiflow/symbiflow-examples/fasm/build/lib.linux-x86_64-3.7/fasm/parser', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/tim/symbiflow/symbiflow-examples/fasm/build/lib.linux-x86_64-3.7/fasm/parser', '-DPYTHON_EXECUTABLE=/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/bin/python', '-DANTLR_RUNTIME_TYPE=static', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
```

litghost, 2021-02-01:
Couple things.  First, we have an PyPi account setup, so if someone adds a PyPi publish to https://github.com/symbiflow/fasm, this issue becomes must easier to solve!

Second, there are instructions in https://github.com/SymbiFlow/fasm/blob/master/README.md on how to build FASM with the ANTLR parser.

tcal-x, 2021-02-01:
Ok, I think it is resolved.   I was able to generate the bitstream without the fallback warning:

```
Incr Criticality updates 0 in 0 sec
Full Criticality updates 1 in 4.2928e-05 sec
Writing Implementation FASM: top.fasm
The entire flow of VPR took 19.0512 seconds.
FASM extra: top_fasm_extra.fasm
writing final fasm
cd build/arty_35 && symbiflow_write_bitstream -d artix7 -f top.fasm -p xc7a35tcsg324-1 -b top.bit
Writing bitstream ...
```

closed 2021-02-01 by tcal-x