chipsalliance/f4pga-examples #122: Nothing for Icestorm?

donn, 2021-01-27:
Hey, despite Symbiflow mentioning ice40 support based on icestorm being supported, there seems to be no mention of it in this repo.

Is this anywhere on the radar?

pgielda, 2021-01-27:
Those are indeed VPR-based Symbiflow examples. Long term, yes I think its on the radar, short term probably not. Same thing goes for Windows and Mac support, its on the radar, in fact we are already packaging parts of the toolchain for Win and Mac, but for now only Linux is being tested/supported.

mithro, 2021-01-27:
@donn - As the icestorm / nextpnr flows already have a lot of good tutorials, we have been focusing on the areas which are still lacking. We would love to see some contributions for them however!