chipsalliance/f4pga-examples #123: nmigen tilegrid.json path issue when using symbiflow toolchain

nickoe, 2021-01-27:
@litghost were very persisten that I should create an issue about this one in this repo, so here it is.

The error in short is:
``` 
FileNotFoundError: [Errno 2] No such file or directory: '/home/nickoe/symbiflow_toolchain_deleteme/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/xc7a35tcsg324-1/tilegrid.json'
```
Essentially it looks like the path to `tilegrid.json` is wrong, it is missing the `artix7` folder that contains `xc7a35tcpg236-1`.

```
Writing Implementation FASM: top.fasm
The entire flow of VPR took 24.8408 seconds.
FASM extra: top_fasm_extra.fasm
writing final fasm
Writing bitstream ...
Traceback (most recent call last):
  File "/home/nickoe/symbiflow_toolchain_deleteme/xc7/conda/envs/xc7/bin/xcfasm", line 8, in <module>
    sys.exit(main())
  File "/home/nickoe/symbiflow_toolchain_deleteme/xc7/conda/envs/xc7/lib/python3.7/site-packages/xc_fasm/xc_fasm.py", line 63, in main
    emit_pudc_b_pullup=args.emit_pudc_b_pullup)
  File "/home/nickoe/symbiflow_toolchain_deleteme/xc7/conda/envs/xc7/lib/python3.7/site-packages/xc_fasm/fasm2frames.py", line 120, in fasm2frames
    assembler = fasm_assembler.FasmAssembler(db)
  File "/home/nickoe/symbiflow_toolchain_deleteme/xc7/conda/envs/xc7/lib/python3.7/site-packages/prjxray/fasm_assembler.py", line 32, in __init__
    self.grid = db.grid()
  File "/home/nickoe/symbiflow_toolchain_deleteme/xc7/conda/envs/xc7/lib/python3.7/site-packages/prjxray/db.py", line 152, in grid
    self._read_tilegrid()
  File "/home/nickoe/symbiflow_toolchain_deleteme/xc7/conda/envs/xc7/lib/python3.7/site-packages/prjxray/db.py", line 133, in _read_tilegrid
    'tilegrid.json')) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/nickoe/symbiflow_toolchain_deleteme/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/xc7a35tcsg324-1/tilegrid.json'
Traceback (most recent call last):
  File "/home/nickoe/symbiflow_toolchain_deleteme/xc7/conda/envs/xc7/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/nickoe/symbiflow_toolchain_deleteme/xc7/conda/envs/xc7/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/tmp/symbiflow-examples_deleteme/nmigen/nmigen-boards/nmigen_boards/arty_a7.py", line 219, in <module>
    ArtyA7Platform(toolchain="Symbiflow").build(Blinky(), do_program=False)
  File "/tmp/symbiflow-examples_deleteme/nmigen/nmigen/build/plat.py", line 99, in build
    products = plan.execute_local(build_dir)
  File "/tmp/symbiflow-examples_deleteme/nmigen/nmigen/build/run.py", line 98, in execute_local
    subprocess.check_call(["sh", "{}.sh".format(self.script)])
  File "/home/nickoe/symbiflow_toolchain_deleteme/xc7/conda/envs/xc7/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', 'build_top.sh']' returned non-zero exit status 1.
```

I am trying to use the symbiflow toolchain with nmigen, and I had to patch a little so it is not stright forward to test, but I made this script to document my steps taken:
```
#!/bin/bash -ex

# Following instructions...
# https://symbiflow-examples.readthedocs.io/en/latest/getting-symbiflow.html

cd /tmp
git clone https://github.com/SymbiFlow/symbiflow-examples symbiflow-examples_deleteme
cd symbiflow-examples_deleteme

export FPGA_FAM=xc7
export INSTALL_DIR=${HOME}/symbiflow_toolchain_deleteme
#mkdir -p ${INSTALL_DIR}/${FPGA_FAM}/
#https://raw.githubusercontent.com/SymbiFlow/symbiflow-examples/master/xc7/environment.yml

wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh -O conda_installer.sh
bash conda_installer.sh -u -b -p $INSTALL_DIR/$FPGA_FAM/conda;
source "$INSTALL_DIR/$FPGA_FAM/conda/etc/profile.d/conda.sh";
conda env create --force -f ${FPGA_FAM}/environment.yml
#conda env create --force -f https://raw.githubusercontent.com/SymbiFlow/symbiflow-examples/master/xc7/environment.yml

mkdir -p $INSTALL_DIR/xc7/install

wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/112/20201208-080919/symbiflow-arch-defs-install-7c1267b7.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/112/20201208-080919/symbiflow-arch-defs-xc7a50t_test-7c1267b7.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/112/20201208-080919/symbiflow-arch-defs-xc7a100t_test-7c1267b7.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/112/20201208-080919/symbiflow-arch-defs-xc7a200t_test-7c1267b7.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/112/20201208-080919/symbiflow-arch-defs-xc7z010_test-7c1267b7.tar.xz | tar -xJC $INSTALL_DIR/xc7/install

conda activate $FPGA_FAM

# cloning forks/fixes of the nmigen stuff
git clone https://github.com/nickoe/nmigen.git -b fixup_symbiflow_toolchain_xilinx_7series
cd nmigen
pip install -e .

git clone https://github.com/nickoe/nmigen-boards.git -b basys3
cd nmigen-boards
pip install -e .

export PATH="$INSTALL_DIR/$FPGA_FAM/install/bin:$PATH";
conda activate $FPGA_FAM

set +x
echo
echo
echo
echo ============================================================
echo Run this:
echo
echo export FPGA_FAM=xc7
echo export INSTALL_DIR=${HOME}/symbiflow_toolchain_deleteme
echo export PATH="$INSTALL_DIR/$FPGA_FAM/install/bin:$PATH"
echo conda activate $FPGA_FAM
echo python3 -m nmigen_boards.arty_a7
echo
echo ============================================================
```

acomodi, 2021-01-28:
Thanks for filing this issue @nickoe!

I could reproduce the error and found the solution. Basically what is missing is the `-d` parameter to [write_bitstream](https://github.com/nickoe/nmigen/blob/b1af42fa046b302545b0fb79fc6e40897f4d540c/nmigen/vendor/xilinx_7series.py#L261-L264) which identifies the FPGA family (artix7, zynq7, kintex7, etc), to get the correct database info.

By adding that, and calling `python3 -m nmigen_boards.arty_a7` once again, I could get to the end of the flow.

nickoe, 2021-01-28:
@acomodi Thank you for investigating, I will try that when I get home and update my PR to nmigen.

tcal-x, 2021-01-28:
@nickoe & @acomodi , this is great!

mithro, 2021-01-28:
There should probably be a much better error messages?

nickoe, 2021-01-28:
@mithro I am not sure anything easy can be done for that, but the `symbiflow_write_bitstream` does not seem to have any help text, so I don't really know what args it takes.

mithro, 2021-01-28:
> @mithro I am not sure anything easy can be done for that, but the `symbiflow_write_bitstream` does not seem to have any help text, so I don't really know what args it takes. Or maybe it is just nmigen that should print the failing command?

@acomodi logged https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1985 -- That should most certainly be fixed!

nickoe, 2021-01-28:
Mmm, I think the is a regression in the symbiflow_write_bitstream, I patched that, but it still errors about the `tilegrid.json` missing. It  appears that this file is only defined for some devices, is this expected?

```
(xc7) [nickoe@x280-arch nmigen_boards]$ ls /home/nickoe/symbiflow_install/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7*
/home/nickoe/symbiflow_install/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a100t:
node_wires.json  tileconn.json  tilegrid.json

/home/nickoe/symbiflow_install/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a100tcsg324-1:
package_pins.csv  part.json  part.yaml

/home/nickoe/symbiflow_install/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a100tfgg676-1:
node_wires.json  package_pins.csv  part.json  part.yaml  tileconn.json  tilegrid.json

/home/nickoe/symbiflow_install/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a200t:
node_wires.json  tileconn.json  tilegrid.json

/home/nickoe/symbiflow_install/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a200tffg1156-1:
package_pins.csv  part.json  part.yaml

/home/nickoe/symbiflow_install/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a200tsbg484-1:
node_wires.json  package_pins.csv  part.json  part.yaml  tileconn.json  tilegrid.json

/home/nickoe/symbiflow_install/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a35tcpg236-1:
package_pins.csv  part.json  part.yaml

/home/nickoe/symbiflow_install/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a35tcsg324-1:
node_wires.json  package_pins.csv  part.json  part.yaml  tileconn.json  tilegrid.json

/home/nickoe/symbiflow_install/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a35tftg256-1:
node_wires.json  package_pins.csv  part.json  part.yaml  tileconn.json  tilegrid.json

/home/nickoe/symbiflow_install/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a50t:
node_wires.json  tileconn.json  tilegrid.json

/home/nickoe/symbiflow_install/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a50tfgg484-1:
node_wires.json  package_pins.csv  part.json  part.yaml  tileconn.json  tilegrid.json
```

nickoe, 2021-01-28:
Ok, the last thing with the tilegrid appears fixed with `conda install -c litex-hub prjxray-db=0.0_0239_gd87c844=20201120_145821`.

nickoe, 2021-01-28:
It appears that the suggestion proposed in https://github.com/SymbiFlow/symbiflow-examples/issues/123#issuecomment-769025271 is not required, but certainly related to the issue. There was a bug in `symbiflow_write_bitstream`, it should find the db root by  itself, there is a proposed patch in:

https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1986

nickoe, 2021-01-31:
This is resolved with this fix in arch defs, https://github.com/nickoe/symbiflow-arch-defs/commit/e16b2095665a5ae390047d6649d3b2d3b44bc333

closed 2021-01-31 by nickoe