chipsalliance/f4pga-examples #124: Modify requirements.txt to accomodate changes in fasm

Mattia9875, 2021-02-19:
In ```xc7/requirements.txt``` 
substitute ```git+https://github.com/symbiflow/fasm``` 
with ```--no-use-pep517 git+https://github.com/SymbiFlow/fasm.git#egg=fasm```

as of [issue](https://github.com/SymbiFlow/fasm/issues/60) over on /symbiflow/fasm

litghost, 2021-02-19:
Now that `fasm` is on PyPi, the xc7/requirements can just remove the `fasm` requirement completely now that it is in PyPi.

Mattia9875, 2021-02-19:
You mean that in the sense that i could do it on my system? or that is the general way to fix it?

litghost, 2021-02-19:
> You mean that in the sense that i could do it on my system? or that is the general way to fix it?

As in you/I could submit a PR to fix this?

Mattia9875, 2021-02-19:
Yes, though Tim responded on the issue I cited saying that he is currently working on in so I don't know how this issue will stand