chipsalliance/f4pga-examples #126: Question about constraints

rodrigomelo9, 2021-02-24:
Hi. It seems that both, `pcf` and `xdc` are supported.

I was comparing the three xc7 examples. Taking for example `arty`, I found:
* `counter_test/arty.xdc`
* `linux_litex_demo`:
    * `arty.pcf`
    * `arty.xcf`
* `picosoc_demo/arty.pcf`

Comparing the `Makefiles`:
* For `symbiflow_synth`, `counter_test` and `linux_litex_demo` specify `-x ${XDC}` (don't used by `picosoc_demo`)
* For `symbiflow_place`, `linux_litex_demo` and `picosoc_demo` specify `-p ${PCF}` (don't used by `counter_test`)

Is the same to use `-x ${XDC}` for `symbiflow_synth` and `-p ${PCF}` for `symbiflow_place`? Do I need to choose only one option? Has it sense to support the two alternatives? Recommendation? In my opinion, in place has more sense (but is using pcf instead of xdc). Is it taken into account for synthesis?).

Clarification will be welcome.

Regards,
Rodrigo

litghost, 2021-02-24:
We are moving to XDC, but some of the old demos were written when only PCF support existed.

rodrigomelo9, 2021-02-24:
Ok, great. What about to specify in `symbiflow_synth` instead of `symbiflow_place`? Is it the final destination? (`symbiflow_synth`?) There, we can specify the timing and place constraints?

mithro, 2021-02-24:
I think that is just a mistake?

mithro, 2021-02-24:
@acomodi / @kgugala -- Thoughts?

acomodi, 2021-02-24:
@rodrigomelo9 It actually is expected it to be like that.

- PCF: this type of constraints is only concerning placement of IOs, therefore it is required during `symbiflow_place` as the placement constraints are read at that step only (and a constraints file tailored for VPR is generated as output)
- XDC: this file can contain different kind of constraints and not only placement ones (e.g. timing, etc). We have a yosys plugin capable of parsing the XDC constraints and save the placement locations of the IOs directly in the eblif, thus not requiring any additional arguments in the `symbiflow_place` step.

Additionally, XDC and PCF should not be used at the same time and it is better to use XDC, given that they can contain also timing and other constraints.

mithro, 2021-02-24:
@acomodi -- This seems incorrect. The XDC should contain the IO placements (as well as other constraints)? Is the problem that we are missing a tool which generates the IO constraints file for VPR from an XDC input?

acomodi, 2021-02-24:
@mithro This is what is happening, the XDC contains all kinds of constraints. The yosys plugin can read them and save them in the eblif, for the io place constraints script to generate the VPR constraint files

mithro, 2021-02-24:
@acomodi -- So the PCF file is being generated by the output of Yosys which is then converted into the correct format needed by VPR?

acomodi, 2021-02-24:
@mithro No. There are two independent flows concerning IO placement constraints, depending on what the user chooses:
XDC flow: yosys parses the file and saves the constraints in the eblif. The placement constraints script will read from the eblif and generate the VPR constraints. No PCF is produced
PCF flow: yosys does nothing. The placement constraints script reads the PCF and generates the VPR constraints. (The eblif is read also here, but only to get the top level IO nets)

mithro, 2021-02-24:
I think we might have discussed this previously. We should hand in the PCF at the same place we hand in the XDC and then the rest of the flow should be the same.

See https://github.com/SymbiFlow/yosys-symbiflow-plugins/issues/30 and maybe https://docs.google.com/drawings/d/1r2LXypJF5AD40LfHegml3_fIvPT2jZ3n2OZYW9-9dLU/edit ?

rodrigomelo9, 2021-02-25:
Thanks. My question was answered, so this issue can be closed if you want (I mean, it seems that @mithro detected another problem, but that is not what this issue was about).

Regards