chipsalliance/f4pga-examples #131: check_if_rest_of_carry4_is_unused:    assert cell_idx < len(O_ports) - 1

tcal-x, 2021-03-20:
I encountered this using the symbiflow-examples installation and attempting to use it to build a LiteX SoC.

I added some debugging in the script; there were 4 O\_ports, and cell\_idx was 3.

I'll attach the testcase as a tarfile here.

tcal-x, 2021-03-20:
```
16. Executing JSON backend.
ERROR: TCL interpreter returned an error: Traceback (most recent call last):
  File "/media/tim/GIT/opt/symbiflow/xc7/install/share/symbiflow/scripts/fix_xc7_carry.py", line 605, in <module>
    main()
  File "/media/tim/GIT/opt/symbiflow/xc7/install/share/symbiflow/scripts/fix_xc7_carry.py", line 598, in main
    design, top_module, bit_to_cells, bit_to_nets, chain
  File "/media/tim/GIT/opt/symbiflow/xc7/install/share/symbiflow/scripts/fix_xc7_carry.py", line 559, in fixup_congested_rows
    check_if_rest_of_carry4_is_unused(cellname, cell_idx + 1):
  File "/media/tim/GIT/opt/symbiflow/xc7/install/share/symbiflow/scripts/fix_xc7_carry.py", line 508, in check_if_rest_of_carry4_is_unused
    assert cell_idx < len(O_ports) - 1
AssertionError
```

tcal-x, 2021-03-21:
[testcase131.tar.gz](https://github.com/SymbiFlow/symbiflow-examples/files/6176692/testcase131.tar.gz)

To run the testcase, enter the SymbiFlow environment with $FPGA_FAM=xc7,  ungzip/untar the test case, and source "source_me" in the directory.

mithro, 2021-03-24:
@litghost / @acomodi - Can you take a look at this?

litghost, 2021-03-26:
I believe this was fixed in arch-defs already? https://github.com/SymbiFlow/symbiflow-arch-defs/commit/61f7d23ff10f386286261b8f07a9885b78cd0b55  That fix landed on January 24.  @acomodi When was symbiflow-examples last updated?  As part of the PCIe work, we need to push a new symbiflow-examples, so maybe good to wait until some more of the GTP stuff lands?

tcal-x, 2021-03-26:
> I believe this was fixed in arch-defs already? [SymbiFlow/symbiflow-arch-defs@61f7d23](https://github.com/SymbiFlow/symbiflow-arch-defs/commit/61f7d23ff10f386286261b8f07a9885b78cd0b55) That fix landed on January 24. @acomodi When was symbiflow-examples last updated? As part of the PCIe work, we need to push a new symbiflow-examples, so maybe good to wait until some more of the GTP stuff lands?

Waiting is fine by me.  I'll come back to this, but now I'm on Nexus stuff.

acomodi, 2021-03-26:
I have opened a [PR to update the packages](https://github.com/SymbiFlow/symbiflow-examples/pull/135) to the latest version.

> so maybe good to wait until some more of the GTP stuff lands?

I think it is better to update them even before GTP stuff lands, as quite some things changed from the last time we updated the packages. As an example, we were still using the non-optimized VTR package.

mithro, 2021-03-26:
@acomodi - What happened to the idea of having a bot auto-update these?

tcal-x, 2021-03-26:
I attempted using SymbiFlow again after I applied the updates in PR #135.   I ran into the following error regarding a missing tilegrid.json, although I probably made a mistake, since the CI for #135 is all green.

```
symbiflow_write_bitstream -d artix7 -f arty.fasm -p xc7a35tcsg324-1 -b arty.bit > /dev/null
Traceback (most recent call last):
  File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/bin/xcfasm", line 8, in <module>
    sys.exit(main())
  File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/xc_fasm/xc_fasm.py", line 63, in main
    emit_pudc_b_pullup=args.emit_pudc_b_pullup)
  File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/xc_fasm/fasm2frames.py", line 120, in fasm2frames
    assembler = fasm_assembler.FasmAssembler(db)
  File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/prjxray/fasm_assembler.py", line 32, in __init__
    self.grid = db.grid()
  File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/prjxray/db.py", line 152, in grid
    self._read_tilegrid()
  File "/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/prjxray/db.py", line 133, in _read_tilegrid
    'tilegrid.json')) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a35tcsg324-1/tilegrid.json'
```

There *is* a tilegrid.json at:
```
/media/tim/GIT/opt/symbiflow/xc7/conda/envs/xc7/share/symbiflow/prjxray-db/artix7/xc7a50t/tilegrid.json
```

...so I have a feeling I'm out of sync regarding the "all-parts" update.

@acomodi , is there something obvious that I missed applying the updates from #135?   I did a "conda env update -f environment.yml", "pip install -r requirements.txt", and executed the "wget ..." commands.

tcal-x, 2021-03-26:
@acomodi , never mind, I should try a clean install before you take a look.

tcal-x, 2021-03-30:
@acomodi , I built my example after installing symbiflow-examples using your updates, and everything worked for me.   I didn't see the original issue, and I didn't see the tilegrid issue that I mentioned.

acomodi, 2021-03-30:
@tcal-x great! I guess we can merge the update packages PR than

tcal-x, 2021-03-30:
I merged the PR with the updates, but yes, auto-updating would be a better solution.

closed 2021-03-30 by tcal-x