chipsalliance/f4pga-examples #137: Missing plugin in yosys installation (conda)

Mattia9875, 2021-04-03:
I will post the full meaage below, in short, when you try and run an example it fails citing a missing plugin:
```
(xc7) root@ff9495062bb1:/symb/tmp/xc7# TARGET="arty_100" make -C counter_test
make: Entering directory '/symb/tmp/xc7/counter_test'
mkdir -p build/arty_100
cd build/arty_100 && symbiflow_synth -t top -v /symb/tmp/xc7/counter_test/counter.v -d artix7 -p xc7a100tcsg324-1 -x /symb/tmp/xc7/counter_test/arty.xdc 2>&1 > /dev/null
ERROR: Can't load module `./selection': /opt/conda/envs/xc7/bin/../share/yosys/plugins/selection.so: cannot open shared object file: No such file or directory
make: *** [Makefile:45: build/arty_100/top.eblif] Error 1
make: Leaving directory '/symb/tmp/xc7/counter_test'
```

kgugala, 2021-04-03:
Hi @Mattia9875 the `selection`  plugin has been renamed to `design_introspection`. I suppose you need to get newer toolchain installation. Can you try a fresh run? 

The CI in this repo is actually testing the guide in nightly builds, so if you start with fresh clone you should be able to build the designs locally.

Mattia9875, 2021-04-03:
Tried a fresh install in a docker container running miniconda3 as base, no luck there, i'll post my dockerfile
```
FROM continuumio/miniconda3

#taking the board model as build time argument
ARG BOARD_MODEL=xc7a100t
WORKDIR /symb
RUN apt update && \
    apt install -y git wget xz-utils xc3sprog

#cloning into symbiflow-examples
RUN git clone https://github.com/SymbiFlow/symbiflow-examples && \
    cd symbiflow-examples

#creating the conda enviroment
RUN conda env create -f ${INSTALL_DIR}/symbiflow-examples/${FPGA_FAM}/environment.yml

#setting the shell inside the enviroment (xc7 -> FAMILY)
SHELL ["conda", "run", "-n", "xc7", "/bin/bash", "-c"]
RUN mkdir -p $INSTALL_DIR/${FPGA_FAM}/install && \
    wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/112/20201208-080919/symbiflow-arch-defs-install-7c1267b7.tar.xz | tar -xJC $INSTALL_DIR/${FPGA_FAM}/install       && \
    wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/112/20201208-080919/symbiflow-arch-defs-${BOARD_MODEL}_test-7c1267b7.tar.xz | tar -xJC $INSTALL_DIR/${FPGA_FAM}/install

#linking the toolchain commands
ENV PATH="$PATH:/symb/${FPGA_FAM}/install/bin"
```
PS: Same dockerfile worked

kgugala, 2021-04-04:
you need to update these lines:

```
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/112/20201208-080919/symbiflow-arch-defs-install-7c1267b7.tar.xz | tar -xJC $INSTALL_DIR/${FPGA_FAM}/install       && \
    wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/112/20201208-080919/symbiflow-arch-defs-${BOARD_MODEL}_test-7c1267b7.tar.xz | tar -xJC $INSTALL_DIR/${FPGA_FAM}/install
```
you need to grab newer package (you can use the one referenced in the docs)