chipsalliance/f4pga-examples #138: conda env create not working in "Getting Symbiflow"

hansfbaier, 2021-04-13:
```
$ export INSTALL_DIR=/opt/symbiflow

/tmp ⌚ 3:24:59
$ export FPGA_FAM=xc7

/tmp ⌚ 3:25:10
$ bash conda_installer.sh -u -b -p $INSTALL_DIR/$FPGA_FAM/conda;
PREFIX=/opt/symbiflow/xc7/conda
Unpacking payload ...
Collecting package metadata (current_repodata.json): done                                                                     
Solving environment: done

## Package Plan ##

  environment location: /opt/symbiflow/xc7/conda

  added / updated specs:
    - _libgcc_mutex==0.1=main
    - brotlipy==0.7.0=py38h27cfd23_1003
    - ca-certificates==2020.10.14=0
    - certifi==2020.6.20=pyhd3eb1b0_3
    - cffi==1.14.3=py38h261ae71_2
    - chardet==3.0.4=py38h06a4308_1003
    - conda-package-handling==1.7.2=py38h03888b9_0
    - conda==4.9.2=py38h06a4308_0
    - cryptography==3.2.1=py38h3c74f83_1
    - idna==2.10=py_0
    - ld_impl_linux-64==2.33.1=h53a641e_7
    - libedit==3.1.20191231=h14c3975_1
    - libffi==3.3=he6710b0_2
    - libgcc-ng==9.1.0=hdf63c60_0
    - libstdcxx-ng==9.1.0=hdf63c60_0
    - ncurses==6.2=he6710b0_1
    - openssl==1.1.1h=h7b6447c_0
    - pip==20.2.4=py38h06a4308_0
    - pycosat==0.6.3=py38h7b6447c_1
    - pycparser==2.20=py_2
    - pyopenssl==19.1.0=pyhd3eb1b0_1
    - pysocks==1.7.1=py38h06a4308_0
    - python==3.8.5=h7579374_1
    - readline==8.0=h7b6447c_0
    - requests==2.24.0=py_0
    - ruamel_yaml==0.15.87=py38h7b6447c_1
    - setuptools==50.3.1=py38h06a4308_1
    - six==1.15.0=py38h06a4308_0
    - sqlite==3.33.0=h62c20be_0
    - tk==8.6.10=hbc83047_0
    - tqdm==4.51.0=pyhd3eb1b0_0
    - urllib3==1.25.11=py_0
    - wheel==0.35.1=pyhd3eb1b0_0
    - xz==5.2.5=h7b6447c_0
    - yaml==0.2.5=h7b6447c_0
    - zlib==1.2.11=h7b6447c_3

The following NEW packages will be INSTALLED:

  _libgcc_mutex      pkgs/main/linux-64::_libgcc_mutex-0.1-main
  brotlipy           pkgs/main/linux-64::brotlipy-0.7.0-py38h27cfd23_1003
  ca-certificates    pkgs/main/linux-64::ca-certificates-2020.10.14-0
  certifi            pkgs/main/noarch::certifi-2020.6.20-pyhd3eb1b0_3
  cffi               pkgs/main/linux-64::cffi-1.14.3-py38h261ae71_2
  chardet            pkgs/main/linux-64::chardet-3.0.4-py38h06a4308_1003
  conda              pkgs/main/linux-64::conda-4.9.2-py38h06a4308_0
  conda-package-han~ pkgs/main/linux-64::conda-package-handling-1.7.2-py38h03888b9_0
  cryptography       pkgs/main/linux-64::cryptography-3.2.1-py38h3c74f83_1
  idna               pkgs/main/noarch::idna-2.10-py_0
  ld_impl_linux-64   pkgs/main/linux-64::ld_impl_linux-64-2.33.1-h53a641e_7
  libedit            pkgs/main/linux-64::libedit-3.1.20191231-h14c3975_1
  libffi             pkgs/main/linux-64::libffi-3.3-he6710b0_2
  libgcc-ng          pkgs/main/linux-64::libgcc-ng-9.1.0-hdf63c60_0
  libstdcxx-ng       pkgs/main/linux-64::libstdcxx-ng-9.1.0-hdf63c60_0
  ncurses            pkgs/main/linux-64::ncurses-6.2-he6710b0_1
  openssl            pkgs/main/linux-64::openssl-1.1.1h-h7b6447c_0
  pip                pkgs/main/linux-64::pip-20.2.4-py38h06a4308_0
  pycosat            pkgs/main/linux-64::pycosat-0.6.3-py38h7b6447c_1
  pycparser          pkgs/main/noarch::pycparser-2.20-py_2
  pyopenssl          pkgs/main/noarch::pyopenssl-19.1.0-pyhd3eb1b0_1
  pysocks            pkgs/main/linux-64::pysocks-1.7.1-py38h06a4308_0
  python             pkgs/main/linux-64::python-3.8.5-h7579374_1
  readline           pkgs/main/linux-64::readline-8.0-h7b6447c_0
  requests           pkgs/main/noarch::requests-2.24.0-py_0
  ruamel_yaml        pkgs/main/linux-64::ruamel_yaml-0.15.87-py38h7b6447c_1
  setuptools         pkgs/main/linux-64::setuptools-50.3.1-py38h06a4308_1
  six                pkgs/main/linux-64::six-1.15.0-py38h06a4308_0
  sqlite             pkgs/main/linux-64::sqlite-3.33.0-h62c20be_0
  tk                 pkgs/main/linux-64::tk-8.6.10-hbc83047_0
  tqdm               pkgs/main/noarch::tqdm-4.51.0-pyhd3eb1b0_0
  urllib3            pkgs/main/noarch::urllib3-1.25.11-py_0
  wheel              pkgs/main/noarch::wheel-0.35.1-pyhd3eb1b0_0
  xz                 pkgs/main/linux-64::xz-5.2.5-h7b6447c_0
  yaml               pkgs/main/linux-64::yaml-0.2.5-h7b6447c_0
  zlib               pkgs/main/linux-64::zlib-1.2.11-h7b6447c_3

Preparing transaction: done
Executing transaction: done
installation finished.

/tmp ⌚ 3:25:38
$ source "$INSTALL_DIR/$FPGA_FAM/conda/etc/profile.d/conda.sh";

/tmp ⌚ 3:25:46
$ conda env create -f $FPGA_FAM/environment.yml

EnvironmentFileNotFound: '/tmp/xc7/environment.yml' file not found

```
maybe it meant to say this, but it does not work as well:
```
$ conda env create -f $INSTALL_DIR/$FPGA_FAM/environment.yml   

EnvironmentFileNotFound: '/opt/symbiflow/xc7/environment.yml' file not found
```

mithro, 2021-04-13:
I transferred this issue to the correct repo? What OS are you on? I assume a RedHat Linux?

mithro, 2021-04-13:
The GitHub Action which is probably closest to your workflow is https://github.com/SymbiFlow/symbiflow-examples/runs/2328966429?check_suite_focus=true

mithro, 2021-04-13:
```
00:25:45 | + export INSTALL_DIR=~/opt/symbiflow
00:25:45 | + export FPGA_FAM=xc7
00:25:45 | + bash conda_installer.sh -u -b -p $INSTALL_DIR/$FPGA_FAM/conda
```

mithro, 2021-04-13:
@hansfbaier Did you follow these steps?
```
git clone https://github.com/SymbiFlow/symbiflow-examples
cd symbiflow-examples
```
?

hansfbaier, 2021-04-13:
> > @hansfbaier - It would be good to figure out why this isn't failing on GitHub Actions?
> > Can you give me a link to that?

> I transferred this issue to the correct repo? What OS are you on? I assume a RedHat Linux?
No, Ubuntu
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.2 LTS"
```

hansfbaier, 2021-04-13:
> @hansfbaier Did you follow these steps?
> 
> ```
> git clone https://github.com/SymbiFlow/symbiflow-examples
> cd symbiflow-examples
> ```
> 
> ?

No, sorry, I skipped reading that and straight went to the headline 'Toolchain installation'.
I was not aware the prerequisites were part of the installation (because I skipped reading those).

mithro, 2021-04-13:
@hansfbaier - I can totally see how you could do that with the current documentation.

mithro, 2021-04-13:
@hansfbaier It is probably better if the documentation was clear that the command was `conda env create -f <symbiflow-examples-repo>/$FPGA_FAM/environment.yml` somehow.

hansfbaier, 2021-04-13:
> @hansfbaier - I can totally see how you could do that with the current documentation.

Yes, it happens when you get up at three in the morning because of trouble sleeping after the neighbors made a heck of a noise :D

mithro, 2021-04-13:
I think your mistake is totally reasonable.

mithro, 2021-04-13:
I'm going to leave this open to solve that issue.

carlosedp, 2021-04-15:
Also it would be important to add to the documentation a step to install the Python modules with:
`pip3 install -r ./${FPGA_FAM}/requirements.txt` as Conda didn't install it on my Docker container.

I keep getting `ModuleNotFoundError: No module named 'simplejson'` and other missing module errors.

The Dockerfile is on: https://gist.github.com/carlosedp/1f9119e0645e842e474ec1a67b5aed7e

carlosedp, 2021-04-15:
Changed my base image to the non Python3 one and also activated it in the shell. Worked fine.

mithro, 2021-04-15:
I'm confused because I'm pretty sure we only support Python 3 and not Python 2?

carlosedp, 2021-04-15:
I believe it makes no difference since Conda installs it's own Python3.
Don't think changing the image made a difference here.
What fixed is moving the environment activation to it's script that is called on container start.