chipsalliance/f4pga-examples #143: Symbiflow build error when using "Linux-on-litex-vexriscv" project from Enjoy Digital

jhu960213, 2021-04-28:
Hello,

I'm currently trying to compile the bit stream for arty a7 35t fpga in the Linux-on-litex-vexriscv project from enjoy digital. 

Here is the link for this project: https://github.com/litex-hub/linux-on-litex-vexriscv 

I have litex and the required dependencies installed already and have already included the symbiflow toolchain path as well. I followed this tutorial to install the symbiflow toolchain: 

https://symbiflow-examples.readthedocs.io/en/latest/getting-symbiflow.html?fbclid=IwAR3mIqvA7yOOGGvh0LAX1PG1GIcueZqQBcVWE_cKJW9UetRjxWnsJ4iZm40#toolchain-installation 

I was able to install it without any errors. 

So in the Linux-on-litex-vex project, at the part of when you are trying to build the bitstream for the targeted fpga platform, I'm getting some build errors with the symbiflow toolchain. 

This is how I invoked the command: ./make.py --board=arty --toolchain=symbiflow --build and before I invoked this command I made sure to activate the conda environment by doing: $ conda activate xc7 for the right dependencies to show up when using the symbiflow toolchain. 

This is the error that I got: 

[image]

I was able to compile it successfully with vivado in the past. I don't really understand what the error is saying... maybe it's how I configured my symbiflow toolchain installation? Or something else  perhaps a dependency issue, etc. Please advise! Thanks 

I am running ubantu 20.04TS

Best,
J.H

jhu960213, 2021-04-28:
@mithro

mithro, 2021-04-29:
@kgugala - Any idea why we haven't seen this issue with the CI system?

I assume the cause is that LiteX have recently added a SLEW property on the SD Card constraints and we don't support that yet?

acomodi, 2021-04-29:
@jhu960213 I believe that IBUF cells do not support any SLEW attribute indeed, and probably Vivado just ignores it.

Here we define the extra attributes for the various IOs: https://github.com/SymbiFlow/symbiflow-arch-defs/blob/ef20acc081b620f3fbbf1d585360564e1ac789bc/xc/xc7/techmap/iobs.v#L1-L17, and the IBUF does not have any SLEW attribute. Same can be seen in the [SelectIO docs](https://www.xilinx.com/support/documentation/user_guides/ug471_7Series_SelectIO.pdf)

jhu960213, 2021-04-29:
Hi @acomodi,

I looked into the docs also and I saw that slew wasn't supported... in that case would it be possible to comment out the slew part in the file location you mentioned: https://github.com/SymbiFlow/symbiflow-arch-defs/blob/ef20acc081b620f3fbbf1d585360564e1ac789bc/xc/xc7/techmap/iobs.v#L1-L17 
? Or how should I go about fixing this issue? 

Thanks!

jhu960213, 2021-04-29:
Oh nvm I saw that the slew was for OBUF, yeah so is there any other way around this?

acomodi, 2021-04-29:
@jhu960213 I believe that, as short term, you may comment out the SLEW property setting in the XDC file. You can find the XDC file in the gateware directory that is generated by LiteX. As a long-term solution we might allow parsing the SLEW property in the flow, which will just be ignored.

jhu960213, 2021-04-29:
So i dug into a bit how litex generates the arty verilog and constraint files from their migen python source code and I commented the part where they specified sdcard in the "soc_capabilities" section in the Arty Class in make.py: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/3244a2882486484c0f80a8becdba1e3fabd3a731/make.py#L62 and seems like my original error the slew attribute thing went away but now its throwing a different error. It says that the "BSCANE2" primitive is not supported. I dug more into the issue and turns out that symbiflow just doesn't support it at the current moment according to this: https://freenode.irclog.whitequark.org/symbiflow/2021-04-03?

Maybe if there is a way or if you guys could help me out as to how to add this BSCANE2 primitive to the symbiflow toolchain that would be awesome! :)  

[image]

jhu960213, 2021-04-29:
@acomodi

acomodi, 2021-04-29:
@jhu960213 Keep in mind that, adding a new primitive to support might take some time, but the various steps are as follows:

1. Document the primitive in prjxray. This means generating a series of minitests to understand how the BSCANE2 primitive works and then a fuzzer to document it.
2. Add support for place and route. This means adding [cells map and sim](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc/xc7/techmap) and the support for generating the physical block and models XML definitions of the architecture. [This PR](https://github.com/SymbiFlow/symbiflow-arch-defs/pull/2053) is an example on this step, but I see that the BSCANE2 primitive is pretty small actually and might require less work to implement.

Unless it is strictly needed for your design, I suggest temporarily remove the BSCANE2 instantiation.