chipsalliance/f4pga-examples #146: symbiflow_pack fails when building the counter_test example

GuzTech, 2021-04-30:
I've installed the SymbiFlow toolchain as described [here](https://symbiflow-examples.readthedocs.io/en/latest/getting-symbiflow.html), but when I want to run the `counter_test` example, I get an error:

```
❯ TARGET="arty_35" make -C counter_test
make: Entering directory '/home/oguz286/Projects/FPGA/symbiflow-examples/xc7/counter_test'
mkdir -p build/arty_35
cd build/arty_35 && symbiflow_synth -t top -v /home/oguz286/Projects/FPGA/symbiflow-examples/xc7/counter_test/counter.v -d artix7 -p xc7a35tcsg324-1 -x /home/oguz286/Projects/FPGA/symbiflow-examples/xc7/counter_test/arty.xdc 2>&1 > /dev/null
cd build/arty_35 && symbiflow_pack -e top.eblif -d xc7a50t_test 2>&1 > /dev/null
make: *** [Makefile:53: build/arty_35/top.net] Error 1
make: Leaving directory '/home/oguz286/Projects/FPGA/symbiflow-examples/xc7/counter_test'
```

I've tried running the `counter` example in the `symbiflow-arch-defs` repository, and that worked fine. I'm on Arch Linux if that's important somehow.

tcal-x, 2021-04-30:
Hi @GuzTech , can you alter the Makefile and rerun (or rerun the last command explicitly), and capture the stdout/stderr to a file instead of having it sent to /dev/null, and post the error and surrounding text here?

GuzTech, 2021-04-30:
Sure! I've removed the `> /dev/null` part of the part that runs symbiflow_pack and now I get this error:

```
❯ TARGET="arty_35" make -C counter_test
make: Entering directory '/home/oguz286/Projects/FPGA/symbiflow-examples/xc7/counter_test'
cd build/arty_35 && symbiflow_pack -e top.eblif -d xc7a50t_test 2>&1
Using default VPR options.
VPR FPGA Placement and Routing.
Version: 8.1.0-dev+a806b1609
Revision: 8.0.0-3445-ga806b1609
Compiled: 2021-03-12T16:36:40
Compiler: GNU 7.3.0 on Linux-4.15.0-1108-azure x86_64
Build Info: Release IPO PGO VTR_ASSERT_LEVEL=2

University of Toronto
verilogtorouting.org
vtr-users@googlegroups.com
This is free open source code under MIT license.

Unexpected command-line argument '@VPR_ARGS@'

usage: vpr architecture circuit [--pack] [--place] [--route] [--analysis] 
       [--disp {on, off}] [--save_graphics {on, off}] 
       [--graphics_commands GRAPHICS_COMMANDS] [-h] [--version] 
       [--device DEVICE_NAME] [-j NUM_WORKERS] [--timing_analysis {on, off}] 
       [--disable_errors DISABLE_ERRORS] [--suppress_warnings SUPPRESS_WARNINGS] 
       [--route_chan_width CHANNEL_WIDTH] [OTHER_OPTIONS ...]
make: *** [Makefile:53: build/arty_35/top.net] Error 1
make: Leaving directory '/home/oguz286/Projects/FPGA/symbiflow-examples/xc7/counter_test'
```

tcal-x, 2021-04-30:
@GuzTech , I need to call in the experts :)   

@kgugala , @acomodi :  `Unexpected command-line argument '@VPR_ARGS@'` -- any ideas?

GuzTech, 2021-04-30:
Ah I found the problem. Because I first tried the `symbiflow-arch-defs` repository, there apparently were some environment variables set that messed things up. I've opened a new terminal and now that step works.

Now I run into the same problem described in https://github.com/SymbiFlow/symbiflow-examples/issues/108 but I'll look into it tomorrow. Thanks for the help :)

closed 2021-04-30 by GuzTech