chipsalliance/f4pga-examples #148: Sybiflow-examples Arty A7 35 build error on ubuntu platform

akshar001, 2021-05-16:
Hello,
We are trying to build an example of counter.

TARGET="arty_35" make -C counter_test

But we are getting this error at final end, let us know if we can get help around it.

We followed this tutorial

https://symbiflow-examples.readthedocs.io/en/latest/getting-symbiflow.html

But we are getting this error at fasm,

Serial number (magic cookie) for the routing is: 182215957
Circuit successfully routed with a channel width factor of 500.
Incr Slack updates 1 in 2.0079e-05 sec
Full Max Req/Worst Slack updates 1 in 6.534e-06 sec
Incr Max Req/Worst Slack updates 0 in 0 sec
Incr Criticality updates 0 in 0 sec
Full Criticality updates 1 in 2.6983e-05 sec
Writing Implementation FASM: top.fasm
/root/opt/symbiflow/xc7/install/bin/vpr_common: line 116: 423874 Killed                  genfasm ${ARCH_DEF} ${EBLIF} --device ${DEVICE_NAME} ${VPR_OPTIONS} --read_rr_graph ${RR_GRAPH} $@
make: *** [Makefile:62: build/arty_35/top.fasm] Error 137
make: *** Deleting file 'build/arty_35/top.fasm'
make: Leaving directory '/home/vegg/Documents/FPGA/symbiflow-examples/xc7/counter_test'
(xc7) root@vegg-HP-Compaq-6200-Pro-SFF-PC:/home/vegg/Documents/FPGA/symbiflow-examples/xc7#

kgugala, 2021-05-16:
hi @akshar001

Error 137 suggest you ran out of memory. What kind of machine did you use to run build the example?

tcal-x, 2021-05-16:
https://github.com/SymbiFlow/symbiflow-examples/issues/119 for reference.

akshar001, 2021-05-17:
6GB,
I have ordered 4GB another ram now!.

akshar001, 2021-05-17:
@kgugala @tcal-x it's fixed. able to run counter_test on my arty board.
I have upgraded the ram and issue is solved.

closed 2021-05-17 by akshar001