chipsalliance/f4pga-examples #153: Error from instantiating MMCME2_BASE primitive

WhiteNinjaZ, 2021-06-14:
I have been testing the symbiflow toolchain on a few designs and have run into an error while trying to run the pong design on [project_f](https://projectf.io/posts/fpga-pong/). When I try to run the design I get this error from the tool chain: 
``Message: Failed to find matching architecture model for 'MMCME2_BASE'``
From what I can tell this is coming from file [clock_gen_480p.sv](https://github.com/projf/projf-explore/blob/master/lib/clock/xc7/clock_gen_480p.sv) trying to instance the MMCME primitive. Does symbiflow offer support for this yet?

WhiteNinjaZ, 2021-06-30:
@acomodi @tmichalak?

acomodi, 2021-06-30:
@WhiteNinjaZ At the moment there is a PR which is soon-to-be-merged and open to add support for the MMCM primitives in [symbiflow-arch-defs](https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1729). For context, symbiflow-arch-defs is the repository that generates the device data used to place&route.

Once that PR is merged and new device data generated by CI, we will be able to use MMCM primitives. In the meantime, unless strictly necessary, you might use `PLLE2_BASE` or `PLLE2_ADV` primitives instead.

WhiteNinjaZ, 2021-06-30:
Thank you for the info.

WhiteNinjaZ, 2021-07-19:
@acomodi Per chance do you know the time frame for when the MMCME2_BASE support in [symbiflow-arch-defs](https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1729) will be done?

mithro, 2021-07-19:
@mkurc-ant

mkurc-ant, 2021-07-20:
@WhiteNinjaZ Hello. I can't tell about the time frame of doing that but adding `MMCME2_BASE` is a matter of writing a Yosys techmap in the same way as it is done for `PLLE2_BASE`. I'd prefer to do that after landing the MMCM PR but if you are lively interested in having the techmap there I can add it.

On the other hand merging https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1729 is more complicated as there is a CI failure not directly related to PLL/MMCM support which require further investigation.