chipsalliance/f4pga-examples #154: /opt/symbiflow/xc7/install/bin/symbiflow_synth: line 93: prjxray-config: command not found

marcmerlin, 2021-06-17:
Hi, I'm following the doc on https://symbiflow-examples.readthedocs.io/en/latest/building-examples.html and the first 2 examples work fine, but litetx is not as clear.

First "Linux LiteX demo" doesn't make it clear if I should be inside the symbiflow-examples git checkout to create all those things done by litex_setup, or elsewhere.
I made a need litex directory. The build then fails with
```
python3 -m litex.soc.software.mkmscimg bios.bin --little
python3 -m litex.soc.software.memusage bios.elf /home/merlin/fpga/litex/litex-boards/litex_boards/targets/build/digilent_arty/software/bios/../include/generated/regions.ld riscv64-unknown-elf

ROM usage: 26.70KiB 	(20.86%)
RAM usage: 1.69KiB 	(21.09%)

make: Leaving directory '/home/merlin/fpga/litex/litex-boards/litex_boards/targets/build/digilent_arty/software/bios'
INFO:SoC:Initializing ROM rom with contents (Size: 0x6ad0).
INFO:SoC:Auto-Resizing ROM rom from 0x20000 to 0x6ad0.
symbiflow_synth -t digilent_arty -v /home/merlin/fpga/litex/pythondata-cpu-vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv.v /home/merlin/fpga/litex/litex-boards/litex_boards/targets/build/digilent_arty/gateware/digilent_arty.v -d artix7 -p xc7a35tcsg324-1 -x digilent_arty.xdc > /dev/null
/opt/symbiflow/xc7/install/bin/symbiflow_synth: line 93: prjxray-config: command not found
make: *** [Makefile:33: digilent_arty.eblif] Error 127
```

I do have this, but it's not in my path:
```
sauron:~/fpga/litex/litex-boards/litex_boards/targets$ find /opt/symbiflow/ | grep prjxray-config
/opt/symbiflow/xc7/conda/pkgs/prjxray-db-0.0_248_g2e51ad3-20210312_125539/bin/prjxray-config
/opt/symbiflow/xc7/conda/envs/xc7/bin/prjxray-config
```
Your code added this to my path: /opt/symbiflow/xc7/conda/condabin:/opt/symbiflow/xc7/install/bin

I had to do this:
sauron:~/fpga/litex/litex-boards/litex_boards/targets$ export PATH="$PATH:$INSTALL_DIR/$FPGA_FAM/conda/envs/$FPGA_FAM/bin/"

but then:
```
symbiflow_synth -t digilent_arty -v /home/merlin/fpga/litex/pythondata-cpu-vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv.v /home/merlin/fpga/litex/litex-boards/litex_boards/targets/build/digilent_arty/gateware/digilent_arty.v -d artix7 -p xc7a35tcsg324-1 -x digilent_arty.xdc > /dev/null
ERROR: No such command: tcl (type 'help' for a command overview)
make: *** [Makefile:33: digilent_arty.eblif] Error 1
sauron:~/fpga/litex/litex-boards/litex_boards/targets$ find /opt/symbiflow/ -type f -name tcl
sauron:~/fpga/litex/litex-boards/litex_boards/targets$ 
```

marcmerlin, 2021-06-18:
as a side note, ./litex_setup.py install is not hermetic, it needed sudo and installed a bunch of stuff in my system python. Not sure if that was the intent.

acomodi, 2021-06-18:
Hi @marcmerlin, I think that the problem here is that you did not source the conda environment, where `prjxray-config` is located.

Another hint that this might be the case is that you required sudo to install litex, which should not happen if it was to be installed in the conda environment.

marcmerlin, 2021-06-18:
@acomodi  thanks for the reply.
there seems to be some include hell going on, I'm pretty sure I had conda included for the previous examples, but now I get this:
```
sauron:~/fpga/litex/litex-boards/litex_boards/targets$ export FPGA_FAM=xc7
sauron:~/fpga/litex/litex-boards/litex_boards/targets$ export PATH="$INSTALL_DIR/$FPGA_FAM/install/bin:$PATH";
sauron:~/fpga/litex/litex-boards/litex_boards/targets$ source "$INSTALL_DIR/$FPGA_FAM/conda/etc/profile.d/conda.sh"
sauron:~/fpga/litex/litex-boards/litex_boards/targets$ conda activate $FPGA_FAM
sauron:~/fpga/litex/litex-boards/litex_boards/targets$ export PATH="$INSTALL_DIR/$FPGA_FAM/install/bin:$PATH";^C
sauron:~/fpga/litex/litex-boards/litex_boards/targets$ ./digilent_arty.py --toolchain symbiflow --cpu-type vexriscv --build
Traceback (most recent call last):
  File "./digilent_arty.py", line 13, in <module>
    from migen import *
ModuleNotFoundError: No module named 'migen'
sauron:~/fpga/litex/litex-boards/litex_boards/targets$ type python3
python3 is /opt/symbiflow/xc7/conda/envs/xc7/bin/python3
```
which seems to say it's using the wrong python again

That said, I see you just removed the entire block I was having issues with
https://github.com/SymbiFlow/symbiflow-examples/pull/111

So, should I just give up on this entire litex checkout that was out of tree and seemed a bit out of place (it wasn't clear why it was there, or how it plugged in with the next step)

marcmerlin, 2021-06-18:
So, can you confirm that 
1) the entire block you just removed in https://github.com/SymbiFlow/symbiflow-examples/commit/dc44e93754f065adf9d851a1696efb3b2e117053
was a wild goose chase, and not necessary, so I can delete it all on my side and forget it ever happened? :)

2) clear up 
https://symbiflow-examples.readthedocs.io/en/latest/building-examples.html#linux-litex-demo a bit?
I sure get it to build and install, and see litex on serial console, but of course, that doesn't include linux. Some basic steps on how to flash a pre-built working linux would be appreciated.
I see that it could also be retrieved over tftp and I could get that working if needed, but that's more error prone, requires ethernet which I have but many don't have anymore, and somehow it seems much simpler to just flash a pre-built linux image if possible :)