chipsalliance/f4pga-examples #170: Symbiflow Breaks When Module Parameters are Passed Single Words of a Multi Bit Array

WhiteNinjaZ, 2021-07-30:
@acomodi @kgugala I have been testing the symbiflow tool chain with designs from several FPGA websites and have encountered a rather perplexing bug within the toolchain. While working on the [Ad Astra](https://projectf.io/posts/fpga-ad-astra/) designs from project F, I encountered a module that generated a faulty bitstream when using symbiflow but worked properly using the Vivado design suite.  I believe this issue is caused by symbiflow's inability to handle passing specific words of a multi bit array into a module. To demonstrate this, I have included a design that passes each word of a 3x4-bit wire `multiBit` between two modules:
 
```
module top(
    input logic [13:0] sw,
    output logic [3:0] led
    );
    logic [3:0] multiBit [0:2];
    
    test m1(sw[3:0], multiBit[0]);
    test m2(sw[7:4], multiBit[1]);
    test m3(sw[11:8], multiBit[2]);

    assign led =
    (sw[13:12] == 0) ? multiBit[0]:
    (sw[13:12] == 1) ? multiBit[1]:
    multiBit[2];

endmodule

```

```
module test(
    input logic [3:0] switch_val,
    output logic [3:0] multiB
    );

    assign multiB = switch_val;

endmodule

```
This design is meant to display the value of 4 switches (sw[3:0], sw[7:4], or sw[11:8]) on the first 4 LEDs of a basys board. The user can choose which switch set to display on the LEDs by toggling switches 12 and 13 on and off (i.e switch 12 high and switch 13 low will display the value of switches 4-7 on the LEDs). Notice that each word of the multi bit array multiBit is passed into the module named `test`. The `test` module simply assigns the value of a 4 bit input to the value of a 4 bit output. 

When I run these modules through symbiflow the toolchain generates a faulty bitstream. Symbiflow does not produce any errors during the flow, but when I toggle different switches on and off nothing happens on the board (no LEDs light). When I run the same design through Vivado, the design works as expected and I am able to display the values of various switches on the LEDs. 

I have tested a few variations of this design, and it appears that symbiflow can handle accessing individual words of a multi dimensional array so long as the array is only accessed and modified within a single module. Passing a 1D arrays of bits between separate modules also works without flaw. From what I can see, symbiflow only breaks down when specific words of a 2D array are passed into another module as a parameter. 

I believe this issue is being generated during the synthesis step. Looking at both the .eblif file and the top_synth.v file created by Yosys it appears that no logic (such as a MUX) is generated by synthesis. If this is indeed a synthesis issue, am I posting this problem in the right place? What is the best place for this issue?  

I have attached the top.eblif file, top_synth.v file, and the XDC I used for the design bellow:

[top.eblif](https://github.com/SymbiFlow/symbiflow-examples/files/6904577/top_eblif.txt)
[basys3.xdc](https://github.com/SymbiFlow/symbiflow-examples/files/6904583/basys3.txt)
[top_synth.v](https://github.com/SymbiFlow/symbiflow-examples/files/6904573/top_synth.txt)

WhiteNinjaZ, 2021-07-30:
Update: Apparently this issue only occurs with System Verilog files. I converted the code above to Verilog by changing all of the "logic" to "wire" and changing the file extension and the design works properly. I have also succeeded in getting more complicated designs in Project F  with this same type of issue working by converting SV->V with [these instructions](https://electronics.stackexchange.com/questions/78834/systemverilog-to-verilog-translation) and converting logic elements into wires or registers depending on there use case.

mithro, 2021-08-02:
@WhiteNinjaZ - You should get these into sv-tests and talk to @hzeller about options here then.

WhiteNinjaZ, 2021-08-12:
@mithro Moved this issue to sv-tests as requested.