chipsalliance/f4pga-examples #183: Symbiflow_synth command not found

shahzaibk23, 2021-09-14:
While executing, I am getting this error:

`/home/user/symbiflow-examples/xc7/counter_test/arty.xdc 2>&1 > /dev/null
/bin/sh: line 1: symbiflow_synth: command not found
make: *** [/home/kraken/symbiflow-examples/xc7/counter_test/../../common/Makefile:53: /home/kraken/symbiflow-examples/xc7/counter_test/build/arty_35/top.eblif] Error 127
make: Leaving directory '/home/kraken/symbiflow-examples/xc7/counter_test'
`

tcal-x, 2021-09-14:
@shahzaibk23 , it is easy to miss the line `conda activate $FPGA_FAM` in the instructions on https://symbiflow-examples.readthedocs.io/en/latest/building-examples.html -- could that be the issue?

shahzaibk23, 2021-09-15:
I did the installation process again and it worked. 
Thank You! :)

tcal-x, 2021-09-15:
@shahzaibk23 great to hear.   Don't hesitate to let us know about any other issues, or spots in the documentation that are not clear.

shahzaibk23, 2021-09-23:
I am getting this error again, I tried re-installing Symbiflow a few times but this error is still there !!

The list of commands I ran for installing Symbiflow is as follows:

```
wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh -O conda_installer.sh
export INSTALL_DIR=~/opt/symbiflow
export FPGA_FAM=xc7
bash conda_installer.sh -u -b -p $INSTALL_DIR/$FPGA_FAM/conda;
source "$INSTALL_DIR/$FPGA_FAM/conda/etc/profile.d/conda.sh";
wget --no-check-certificate --directory-prefix=$FPGA_FAM --content-disposition https://raw.githubusercontent.com/SymbiFlow/symbiflow-examples/master/xc7/environment.yml
wget --no-check-certificate --directory-prefix=$FPGA_FAM --content-disposition https://raw.githubusercontent.com/SymbiFlow/symbiflow-examples/master/xc7/requirements.txt
conda env create -f $FPGA_FAM/environment.yml
mkdir -p $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/201/20210325-000253/symbiflow-arch-defs-install-1c7a3d1e.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/201/20210325-000253/symbiflow-arch-defs-xc7a50t_test-1c7a3d1e.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/201/20210325-000253/symbiflow-arch-defs-xc7a100t_test-1c7a3d1e.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/201/20210325-000253/symbiflow-arch-defs-xc7a200t_test-1c7a3d1e.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/201/20210325-000253/symbiflow-arch-defs-xc7z010_test-1c7a3d1e.tar.xz | tar -xJC $INSTALL_DIR/xc7/install

```

kgugala, 2021-09-23:
@shahzaibk23 I don't see `conda activate $FPGA_FAM` in your log

shahzaibk23, 2021-09-24:
Yes I ran `conda activate $FPGA_FAM` and after that `TARGET="arty_35" make -C counter_test` 
and the terminal shows me this:

```TARGET="arty_35" make -C counter_test```

mithro, 2021-09-24:
@shahzaibk23 -- Can you please paste the complete output as you typed things into the terminal. It is clear that what you are saying you are doing and what you are actually doing is not the same. It is impossible to debug what is happening with the information you are providing.

shahzaibk23, 2021-09-25:
Hope this might help 
```bash
skashif@HP-Z800-Workstation:~/symbiflow-examples/xc7$ conda activate xc7
(xc7) skashif@HP-Z800-Workstation:~/symbiflow-examples/xc7$ TARGET="arty_35" make -C counter_test
make: Entering directory '/home/skashif/symbiflow-examples/xc7/counter_test'
cd /home/skashif/symbiflow-examples/xc7/counter_test/build/arty_35 && symbiflow_synth -t top -v /home/skashif/symbiflow-examples/xc7/counter_test/counter.v -d artix7 -p xc7a35tcsg324-1 -x /home/skashif/symbiflow-examples/xc7/counter_test/arty.xdc 2>&1 > /dev/null
/bin/sh: 1: symbiflow_synth: not found
make: *** [/home/skashif/symbiflow-examples/xc7/counter_test/../../common/Makefile:53: /home/skashif/symbiflow-examples/xc7/counter_test/build/arty_35/top.eblif] Error 127
make: Leaving directory '/home/skashif/symbiflow-examples/xc7/counter_test'
```

kgugala, 2021-09-27:
can you paste complete installation log? the snippet you pasted suggests that there was something wrong with installation

shahzaibk23, 2021-09-27:
```bash
wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh -O conda_installer.sh
export INSTALL_DIR=~/opt/symbiflow
export FPGA_FAM=xc7
bash conda_installer.sh -u -b -p $INSTALL_DIR/$FPGA_FAM/conda;
source "$INSTALL_DIR/$FPGA_FAM/conda/etc/profile.d/conda.sh";
wget --no-check-certificate --directory-prefix=$FPGA_FAM --content-disposition https://raw.githubusercontent.com/SymbiFlow/symbiflow-examples/master/xc7/environment.yml
wget --no-check-certificate --directory-prefix=$FPGA_FAM --content-disposition https://raw.githubusercontent.com/SymbiFlow/symbiflow-examples/master/xc7/requirements.txt
conda env create -f $FPGA_FAM/environment.yml
mkdir -p $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/201/20210325-000253/symbiflow-arch-defs-install-1c7a3d1e.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/201/20210325-000253/symbiflow-arch-defs-xc7a50t_test-1c7a3d1e.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/201/20210325-000253/symbiflow-arch-defs-xc7a100t_test-1c7a3d1e.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/201/20210325-000253/symbiflow-arch-defs-xc7a200t_test-1c7a3d1e.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/201/20210325-000253/symbiflow-arch-defs-xc7z010_test-1c7a3d1e.tar.xz | tar -xJC $INSTALL_DIR/xc7/install
conda activate $FPGA_FAM
TARGET="arty_35" make -C counter_test
```
After running the last command, I am getting this error:
```bash
(xc7) skashif@HP-Z800-Workstation:~/symbiflow-examples/xc7$ TARGET="arty_35" make -C counter_test
make: Entering directory '/home/skashif/symbiflow-examples/xc7/counter_test'
cd /home/skashif/symbiflow-examples/xc7/counter_test/build/arty_35 && symbiflow_synth -t top -v /home/skashif/symbiflow-examples/xc7/counter_test/counter.v -d artix7 -p xc7a35tcsg324-1 -x /home/skashif/symbiflow-examples/xc7/counter_test/arty.xdc 2>&1 > /dev/null
/bin/sh: 1: symbiflow_synth: not found
make: *** [/home/skashif/symbiflow-examples/xc7/counter_test/../../common/Makefile:53: /home/skashif/symbiflow-examples/xc7/counter_test/build/arty_35/top.eblif] Error 127
make: Leaving directory '/home/skashif/symbiflow-examples/xc7/counter_test'
```

acomodi, 2021-09-27:
Hi @shahzaibk23, I think you ar missing the path exporting step 

```
export PATH="$INSTALL_DIR/$FPGA_FAM/install/bin:$PATH"
```

closed 2021-10-05 by shahzaibk23