chipsalliance/f4pga-examples #186: Unable to import fast Antlr4 parser implementation

ssmolov, 2021-09-30:
Hello, I've tried to walk through "Building examples" tutorial for SymbiFlow, and, when running "TARGET="zybo" make -C counter_test" command to create bitstream for counter example, I get the following error:
```
writing final fasm
cd /home/ssedai/tools/symbiflow-examples/xc7/counter_test/build/zybo && symbiflow_write_bitstream -d zynq7 -f top.fasm -p xc7z010clg400-1 -b top.bit
Writing bitstream ...
/home/ssedai/tools/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/fasm/parser/__init__.py:33: RuntimeWarning: Unable to import fast Antlr4 parser implementation.
  ImportError: cannot import name 'antlr_to_tuple' from 'fasm.parser' (/home/ssedai/tools/symbiflow/xc7/conda/envs/xc7/lib/python3.7/site-packages/fasm/parser/__init__.py)

  Falling back to the much slower pure Python textX based parser
  implementation.

  Getting the faster antlr parser can normally be done by installing the
  required dependencies and then reinstalling the fasm package with:
    pip uninstall
    pip install -v fasm

  """.format(e), RuntimeWarning)
make: Leaving directory '/home/ssedai/tools/symbiflow-examples/xc7/counter_test'
```
I've reinstalled fasm package, but it doesn't help.

acomodi, 2021-09-30:
Hi @ssmolov, even though there are still some problems when it comes to installing the fasm package, the above is not really an error, but rather a warning.

What happens is that `antlr` cannot adopted, and the `textx` library is used instead, so the bitstream is still getting generated.

mithro, 2021-09-30:
Can you get this documented?