chipsalliance/f4pga-examples #187: Conda env setup slow to download packages

alanvgreen, 2021-10-05:
I am following the instructions at https://symbiflow-examples.readthedocs.io/en/latest/getting-symbiflow.html, and am at this step

```
conda env create -f $INSTALL_DIR/xc7/install/environment.yml
```

I have been downloading packages for at least 4 hours so far. Some packages are very fast to download. Others are very slow. The current package, vtr-optimized-8.0.0_ has been downloading for over an hour and is stuck at 0% and 63.6MB.

alanvgreen, 2021-10-05:
Restarting `conda env create` the fourth time caused the vtr package to finish installing (it jumped immediately to 100%) but then I got this error:

```
Collecting package metadata (repodata.json): done
Solving environment: done
Installing pip dependencies: \ Ran pip subprocess with arguments:
['/home/avg/opt/symbiflow/xc7/conda/envs/symbiflow_arch_def_base/bin/python', '-m', 'pip', 'install', '-U', '-r', '/home/avg/opt/symbiflow/xc7/install/condaenv.qwo98bll.requirements.txt']
Pip subprocess output:

Pip subprocess error:
ERROR: Could not open requirements file: [Errno 2] No such file or directory: '/home/avg/opt/symbiflow/xc7/install/requirements.txt'

failed

CondaEnvException: Pip failed
```

It may be that repeated restarts broke conda env create. I will remove symbiflow entirely and restart the installation.

mithro, 2021-10-05:
@kgugala - Can you have someone look into what is going on here?

mithro, 2021-10-05:
It should be investigated why the CI is not seeing these issues?

kgugala, 2021-10-05:
I just ran the whole flow on my machine and package installation took ~5 minutes. @alanvgreen can you try running this again? Maybe it's some kind of network issue on your side?

tcal-x, 2021-10-05:
My 2 cents, I also have no issues -- my experience is like @kgugala 's.

mithro, 2021-10-05:
We should figure out if there is something strange about @alanvgreen 's setup or if it has something to do with @alanvgreen being in Australia.

alanvgreen, 2021-10-05:
On the second go through, it only took three retries. The gcc package failed once, then had then incorrect hash, and worked the thrird time. Vpr timed out twice before succeeding on the third try.

Things that might be strange about my set up
- I am in Australia
- I am using a VM inside ChromeOS

I'll delete it and try again.

alanvgreen, 2021-10-06:
I deleted my ~/opt and ~/.cache directories and tried again. It finished in 3m17s.

Perhaps it was just a CDN having a bad day

mithro, 2021-10-06:
Just some useful links in case I find this issue again;
 * https://github.com/DanielBok/private-conda-repo
 * https://conda.io/projects/conda/en/latest/user-guide/tasks/create-custom-channels.html