chipsalliance/f4pga-examples #188: ERROR: Module $paramod\FDRE\INIT=1'x contains processes, which are not supported by JSON backend (run `proc` first)

Talha-Ahmed-1, 2021-10-05:
Can anyone help me with this.

```bash
TARGET="arty_35" make -C fpga
make[1]: Entering directory '/home/talha/picofoxy/fpga'
mkdir -p build/arty_35
cd build/arty_35 && symbiflow_synth -t Picofoxy -v /home/talha/picofoxy/fpga/Picofoxy.v /home/talha/picofoxy/fpga/PLL_8MHz.v /home/talha/picofoxy/fpga/clk_wiz_0_clk_wiz.v   -d artix7 -p xc7a35tcsg324-1 -x ~/picofoxy/fpga/arty.xdc 2>&1 > /dev/null
ERROR: Module $paramod\FDRE\INIT=1'x contains processes, which are not supported by JSON backend (run `proc` first).
make[1]: *** [Makefile:53: build/arty_35/Picofoxy.eblif] Error 1
make[1]: Leaving directory '/home/talha/picofoxy/fpga'
make: *** [Makefile:22: bitstream] Error 2
```

acomodi, 2021-10-05:
Hi @Talha-Ahmed-1, I believe that the symbiflow packages you have may be out-dated. Can you please try and re-install the packages (https://symbiflow-examples.readthedocs.io/en/latest/getting-symbiflow.html#toolchain-installation)? 

In particular, the error seems to be due to the fact that the processes are not getting cleared before running the JSON backend in Yosys, which does not allow for any processes. The proc clean step has been added to the synth script and should be present in the latest packages.

Talha-Ahmed-1, 2021-10-05:
Thanks it works for me!

closed 2021-10-05 by Talha-Ahmed-1