chipsalliance/f4pga-examples #204: Switch to Symbiflow for Arty?

alanvgreen, 2021-10-21:
Is it time to switch to CFU Playground to Symbiflow as the recommended tool for Arty A7?

kgugala, 2021-12-14:
Using SymbiFlow in CFU playground is great idea - we'll be able to run tests in public CI. As for the wget - I don't see a problem with creating a meta-package which will do wget in its installation script.

tcal-x, 2021-12-14:
We do now have a make target that does both the `wget`s and installs the Symbiflow conda packages, and it runs in CI.

https://github.com/google/CFU-Playground/blob/main/Makefile#L38-L55

https://github.com/google/CFU-Playground/blob/main/.github/workflows/symbiflow.yml 

It is not default -- it's not installed automatically when you do the regular setup.