chipsalliance/f4pga-examples #211: Weird issue with /usr/bin/env on symbiflow-examples runners

mithro, 2021-11-02:
There was a weird failure @ https://github.com/SymbiFlow/symbiflow-examples/runs/4083446723?check_suite_focus=true
```
Run actions/upload-artifact@v2
19:37:55 | bash: line 1: /usr/bin/env: No such file or directory
Error: Process completed with exit code 127.
0s
19:37:55 | Post job cleanup.
19:37:55 | bash: line 1: /usr/bin/env: No such file or directory
Error: Process completed with exit code 127.
```

mithro, 2021-11-02:
FYI -- @kgugala / @tmichalak

mithro, 2021-11-03:
Do you think this is the pre-emption issue?

pgielda, 2021-11-03:
No, this is just some kind of a bug. Happens always on this particular job (centos7 xc7 counter)

mithro, 2021-11-03:
Seems weird.

pgielda, 2021-11-03:
I agree, it is super weird. But its on the node "javacript" actions that we've support for recently... so there must be some kind of an edge case where it does not work. Same runners work fine in multiple projects in CHIPS Alliance org.

AdamOlech, 2021-11-05:
We seem to have fixed the issue. It was most likely related to an erroneous container invocation in the runner code - see https://github.com/antmicro/runner/commit/838478f10244bfd05576c28d38464ac8731e8cc6

kgugala, 2021-11-05:
This issue doesn't seem to be hitting us anymore. I'm closing the issue, will reopen if it happens again.

closed 2021-11-05 by kgugala