chipsalliance/f4pga-examples #218: Command name collision in symbiflow/scripts/xc7/synth.tcl

mithro, 2021-11-29:
From https://gist.github.com/regymm/3cfaefb1cc5922d72a8d1ccc775efddf
```
1. Executing Verilog-2005 frontend: ../cache_cpu.v
Parsing Verilog input from `../cache_cpu.v' to AST representation.
Generating RTLIL representation for module `\cache_cpu'.
Generating RTLIL representation for module `\cacheway'.
Successfully finished Verilog frontend.

-- Running command `tcl /home/petergu/opt/symbiflow/xc7/install/share/symbiflow/scripts/xc7/synth.tcl' --
[TCL: yosys -import] Command name collision: found pre-existing command `cd' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `eval' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `exec' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `read' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.
Loaded SDC plugin
[TCL: yosys -import] Command name collision: found pre-existing command `abc' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `abc9' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `abc9_exe' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `abc9_ops' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `add' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `aigmap' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `alumacc' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `anlogic_eqn' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `anlogic_fixcarry' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `assertpmux' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `async2sync' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `attrmap' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `attrmvcp' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `autoname' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `blackbox' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `bugpoint' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `cd' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `check' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `chformal' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `chparam' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `chtype' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `clean' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `clk2fflogic' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `clkbufmap' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `connect' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `connect_rpc' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `connwrappers' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `coolrunner2_fixup' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `coolrunner2_sop' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `copy' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `cover' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `cutpoint' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `debug' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `delete' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `deminout' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `design' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `dffinit' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `dfflegalize' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `dfflibmap' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `dffunmap' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `dump' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `echo' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `ecp5_gsr' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `edgetypes' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `efinix_fixcarry' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `equiv_add' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `equiv_induct' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `equiv_make' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `equiv_mark' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `equiv_miter' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `equiv_opt' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `equiv_purge' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `equiv_remove' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `equiv_simple' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `equiv_status' -> skip.
[... 150 lines cut ...]
[TCL: yosys -import] Command name collision: found pre-existing command `wbflip' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `wreduce' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_aiger' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_blif' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_btor' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_cxxrtl' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_edif' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_file' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_firrtl' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_ilang' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_intersynth' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_json' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_rtlil' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_simplec' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_smt2' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_smv' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_spice' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_table' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_verilog' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `write_xaiger' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `xilinx_dffopt' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `xilinx_dsp' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `xilinx_srl' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `zinit' -> skip.
```

mithro, 2021-11-29:
@acomodi / @kgugala / @tmichalak - Does this have something to do with the symbiflow plugins or?

tmichalak, 2021-11-29:
These messages are not symbiflow plugins related. They are coming from the `yosys -import` command that is ingesting all the commands such as `plugin -import` or the symbiflow plugins commands into the TCL interpreter.