chipsalliance/f4pga-examples #230: Running Symbiflow-examples with surelog

nelsobe, 2021-12-17:
@acomodi @mkurc-ant In a recent conversation @mithro suggested there was a way to run the symbiflow-examples toolchain using the surelog SystemVerilog front end.  We are preparing to have students run class projects through the tool flow and would like this to be an option.  Is there documentation anywhere on how to do this?  If not, would be happy to create some if we could learn how to do it.

pgielda, 2021-12-21:
@tgorochowik @kgugala

kamilrakoczy, 2021-12-23:
Hello @nelsobe,

Yes, it is possible to use Surelog/UHDM front-end using symbiflow-toolchain, but it requires a few changes in the ``symbiflow_synth`` wrapper.

[Here](https://github.com/antmicro/yosys-uhdm-plugin-integration/pull/271), you can find an example of building Ibex using symbiflow toolchain with Surelog.

The main changes are in the ``symbiflow_synth`` file. I'm adding there an extra argument ``-s|--surelog`` that accepts arguments and passes them to Surelog. You can find more info about possible arguments in the Surelog repository: https://github.com/chipsalliance/Surelog

Basic documentation on how to use ``uhdm-plugin`` can be found here: https://github.com/antmicro/yosys-uhdm-plugin-integration#yosys-uhdm-plugin-integration, but it focuses only on the usage of ``uhdm-plugin`` with yosys and not in the context of symbiflow toolchain.

pgielda, 2021-12-23:
@kamilrakoczy @kgugala it would be great to add an example to symbiflow-examples that shows how to do that, and possibly extend the wrapper scripts to support that out of the box...

kgugala, 2021-12-23:
agree, we actually started working on that

nelsobe, 2022-01-11:
@kgugala 
In our work on running collections of class designs we are finding problems with the existing build in terms of handling certain SystemVerilog features in student designs (things which Vivado handles just fine).  Surelog would be an interesting alternative to add to the flow to see if it handles these features better.

Could provide any details on your progress on working on supporting Surelog?  

And, with the proper direction, my students could work on it - I have a few students (undergrads) who I could put on it right now.  What I think they (and I) would need is someone to point them at what has been done and summarize what needs to be done...

kamilrakoczy, 2022-01-13:
I've prepared draft PR adapting symbiflow-examples to use uhdm-plugin: https://github.com/SymbiFlow/symbiflow-examples/pull/244.

As PR adapting ``symbiflow_synth`` wrapper is [not merged yet](https://github.com/SymbiFlow/symbiflow-arch-defs/pull/2364), in this PR I'm copying locally added ``symbiflow_synth`` to ``$INSTALL_DIR/xc7/install/bin/symbiflow_synth``. Then, setting ``SURELOG_CMD`` to (or any other proper surelog parameters) ``-parse -DSYNTHESIS`` when building examples, switches to using uhdm-plugin as frontend in yosys.

Not all of the symbiflow-examples are working yet, but bunch of them is already working.

Example build cmd: ``cd xc7/ && TARGET="arty_35" SURELOG_CMD="-parse -DSYNTHESIS" make -C counter_test/``. 

Please let us know if you have any other questions.

nelsobe, 2022-04-01:
@kamilrakoczy @tmichalak Has any of this been merged into f4pga-examples so my students can use it "out of the box"?

kamilrakoczy, 2022-04-05:
@nelsobe

PR adding optional (for now) ``systemverilog`` (renamed ``uhdm``) plugin support in f4pga-examples just got merged: https://github.com/chipsalliance/f4pga-examples/pull/244.

Now systemverilog frontend can be used out-of-box, but it is still optional, as not all examples are working yet. Progress on adapting remaining examples can be tracked here: https://github.com/chipsalliance/yosys-f4pga-plugins/issues/287

When updating local environment, please make sure to update both f4pga-arch-defs and conda environment. Example build command didn't change: ``cd xc7/ && TARGET="arty_35" SURELOG_CMD="-parse -DSYNTHESIS" make -C counter_test/``

The old (default yosys verilog) frontend still can be used when ``SURELOG_CMD`` environment is empty.