chipsalliance/f4pga-examples #231: symbiflow-examples also doesn't download the latest prjxray-db

MidsummerNight, 2021-12-20:
Coming from SymbiFlow/symbiflow-arch-defs#2351, I found that `symbiflow-examples` also doesn't download the latest `prjxray-db`. The immediate cause is [environment.yml](https://github.com/SymbiFlow/symbiflow-examples/blob/6ce90115fae2bb827f11efe02fb44297064b458d/xc7/environment.yml) not pointing to the latest `prjxray-db` version (is there a bot to regularly update this file?), but could there be conda scripts lacking the `spartan7` parameter as well? @acomodi

umarcor, 2021-12-30:
Ref: #223.

kgugala, 2021-12-31:
@MidsummerNight as discussed in #223 we don't want to use latest here - the idea is to have stable examples acting as entry point for users.

We still missing architecture definition for spartan7 so bumping DB would not really add S7 support to example designs

MidsummerNight, 2021-12-31:
@kgugala Understood, then I'll close this issue since spartan7 doesn't seem to be able to make it here any time soon.

closed 2021-12-31 by MidsummerNight