chipsalliance/f4pga-examples #239: udev permissions: add info to docs

tcal-x, 2022-01-05:
The user's first experience with an FPGA board might be with this repo, so they may encounter the usual udev permission issues while doing the examples here.   Add the info about how to update udev rules for the supported boards.

fyi @betocool-prog

pgielda, 2022-01-05:
I would say this is both "jtag" specific (you need permissions to access a usb device if your JTAG is a usb device) and also distro specific (you can have a distro without udev), and over all its a general problem with using anything that you connect over USB when you're not running as root. So the question is if we should really be covering this.
I guess in practice this is usually related to using openocd so maybe we should point to https://openocd.org/doc/html/Running.html where its stated that "On Linux, this usually involves installing a file in /etc/udev/rules.d, so OpenOCD has permissions."

pgielda, 2022-01-05:
@kgugala