chipsalliance/f4pga-examples #240: What is a suitable editor for Symbiflow projects?

betocool-prog, 2022-01-10:
I'm working through the Symbiflow examples and modifying some using VS Code with the "Verilog-HDL/SystemVerilog/Bluespec SystemVerilog" plugin. So far I've seen, it does a good job at highlighting syntax correctly, but that's about it. I have not been able to configure it to be able to jump between modules (e.g.: as with a C++ extension), which I'm missing.

I do like the attached terminal on VS Code, 

What alternatives IDE's are there to VS Code, or alternative extensions to VS Code that work well?

Cheers,

Alberto

pgielda, 2022-01-10:
Symbiflow is really a set of command line tools that take HDL language (text files, e.g. Verilog) to "compile" them (synthesize, place & route, generate fasm and finally generate an FPGA bitstream from it), very much (spiritually) like gcc or llvm (that take e.g. C code and spit out a binary). Therefore IDE is the programmers choice.

VS Code is definitely one option. I personally prefer old-school console tools like vim + console.

betocool-prog, 2022-01-12:
Unfortunately I never went much into vi(m) unless absolutely necessary. I don't mind the console aspect of the synthesis, placement and routing process.

As projects get larger and you have more and more modules, I find that text only editors lack the ability to click and jump between them. Not a deal breaker, but a nice to have.

pgielda, 2022-01-12:
what I am trying to say is that the toolchain is not limiting your ability to choose anything you want. So anything that works for you is suitable. There is no technical limitation in Symbiflow (to my knowledge) that prevents you from using any code editor you want.

closed 2022-01-12 by pgielda