chipsalliance/f4pga-examples #241: Strip spaces from variables TOP and TARGET in common/common.mk

nelsobe, 2022-01-11:
@kgugala @acomodi @mkurc-ant 
Hi Guys, my students are running a Makefile adapted from the one in symbiflow examples to compile their designs as a part of a project to run tons of designs through the flow.  One of them inadvertantly put a trailing space on the line that says 

"TOP := top "  

That caused common.mk to throw a whole bunch of errors and the error messages made no sense.  The problem is that common.mk forms filenames like ${TOP}.eblif and the like and the space is thus a problem.

My solution is to add these lines at the top of common.mk:

TOP := $(strip ${TOP})
TARGET := $(strip ${TARGET})

to remove the spaces.  Before doing so and doing a PR I thought I would ask for your input.   Thoughts?

You could say people shouldn't do things like add extra spaces but they will and so sanitizing the input seems like a reasonable thing to do.

mkurc-ant, 2022-01-12:
Hello! Yes, that's reasonable. Making tools for users more robust and forgiving when provided inaccurate input is always welcome. Please go ahead and make the PR.

nelsobe, 2022-01-12:
Done as PR #243.

closed 2022-01-12 by nelsobe