chipsalliance/f4pga-examples #245: litex produces non-functional bitfile for digilent cmod a7 35t with symbiflow

bl0x, 2022-01-19:
What is observed:
* A compiled litex bitfile for CMOD A7 35t using symbiflow toolchain does not work
    * LED chaser does not blink the LEDs
    * logging in via lxterm is not possible
* On the other hand, a bitfile compiled with Vivado works just as expected

What is expected:
* The bitfile should work also when compiled using symbiflow toolchain

Prerequisites:
* Symbiflow is installed according to the [instructions](https://symbiflow-examples.readthedocs.io/en/latest/getting-symbiflow.html) on the web
* I'm using this to set up my environment:
```
export INSTALL_DIR=~/opt/symbiflow
export FPGA_FAM="xc7"
export PATH="$INSTALL_DIR/$FPGA_FAM/install/bin:$PATH";
source "$INSTALL_DIR/$FPGA_FAM/conda/etc/profile.d/conda.sh"
conda activate $FPGA_FAM
```
* Litex is installed according to the [quick start guide](https://github.com/enjoy-digital/litex#quick-start-guide)

Steps to reproduce:
* Build litex 
```
cd /base/dir/of/litex
python -m litex_boards.targets.digilent_cmod_a7 --toolchain symbiflow --build
```
* Upload resulting bitfile to CMOD A7 module with openOCD
```
openocd -f digilent_cmod_a7.cfg -c "init; pld load 0 digilent_cmod_a7.bit; exit;"
```

Versions:
```
yosys:
Yosys 0.9+4270 (git sha1 539d4ee9, x86_64-conda_cos6-linux-gnu-gcc 1.24.0.133_b0863d8_dirty -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -fdebug-prefix-map=/home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/yosys_1628927583200/work=/usr/local/src/conda/yosys-0.9_5567_g539d4ee9 -fdebug-prefix-map=/home/bloeher/opt/symbiflow/xc7/conda/envs/xc7=/usr/local/src/conda-prefix -fPIC -Os -fno-merge-constants)

vpr:
VPR FPGA Placement and Routing.
Version: 8.1.0-dev+06317d042
Revision: 8.0.0-4118-g06317d042
Compiled: 2021-08-13T07:42:12
Compiler: GNU 9.3.0 on Linux-4.15.0-1113-azure x86_64
Build Info: Release IPO PGO VTR_ASSERT_LEVEL=2

symbiflow-arch-defs:
Version: 20211116-000105

litex (installed on 20220117):
litedram: 24fad45 frontend/dma/LiteDRAMDMAReader: Make sure to flush FIFO/Reservation counter when disabled.
liteeth: 14edaae frontend/stream: Add data_width support.
litehyperbus: a950fa7 Bump year.
liteiclink: 7718e96 Bump year.
litejesd204b: 0c90b3e Bump year.
litepcie: 5446109 tlp/(de)packetizer: Cosmetic cleanup.
litesata: c068dc8 Bump year.
litescope: 42a3577 ci: Install ninja-build/meson.
litesdcard: e52c731 Bump year.
litespi: 4a8e149 Bump year.
litex: b9584531 soc/add_video_framebuffer: Use clock_faster_than_sys when equal.
litex-boards: 834125e platforms/digilent_nexys_video: add missing CD pin
migen: ac70301 sayma_amc2: add SSP1 pins
```

bl0x, 2022-01-21:
I've tried the same build with litex and symbiflow environment set up by [CFU-playground](https://github.com/google/CFU-Playground).

This works!

Versions used in CFU-playground

```
yosys:
Yosys 0.13+3 (git sha1 61324cf55, x86_64-conda_cos6-linux-gnu-gcc 1.24.0.133_b0863d8_dirty -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -fdebug-prefix-map=/home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/yosys_1642201854658/work=/usr/local/src/conda/yosys-0.13_4_g61324cf55 -fdebug-prefix-map=/d/land6/bloeher/CFU-Playground/env/conda/envs/cfu-common=/usr/local/src/conda-prefix -fPIC -Os -fno-merge-constants)

vpr:
VPR FPGA Placement and Routing.
Version: 8.1.0-dev+116f30cb8
Revision: 8.0.0-5105-g116f30cb8
Compiled: 2022-01-14T23:42:34
Compiler: GNU 9.3.0 on Linux-5.11.0-1025-azure x86_64
Build Info: Release IPO PGO VTR_ASSERT_LEVEL=2

symbiflow-arch-defs:
version: 20211222-000718

litex:
$ for d in lite* migen nmigen; do echo -n "$d: "; git -C $d log --pretty=format:"%h%x09%an%x09%ad%x09%s" | head -n1 ; done |sort
litedram: 17c19de       Florent Kermarrec Mon Nov 29 13:22:11 2021 +0100        frontend/dma: Move ack of write responses and other cosmetic cleanups.
liteeth: 7acb2a8        enjoy-digital Sun Nov 21 21:19:17 2021 +0100    Merge pull request #96 from lschuermann/dev/compliant-xgmii-fixup
litehyperbus: 9f118e8   Florent Kermarrec Tue Aug 31 17:50:01 2021 +0200        Copyrights: Bump year.
liteiclink: 3d8ecdb     Florent Kermarrec Thu Apr 1 18:59:33 2021 +0200 serdes/gtx_7series: Set TXDIFFCTRL default control value also to 0b1000 (800mV).
litepcie: 90013df       enjoy-digital Tue Nov 30 09:47:42 2021 +0100    Merge pull request #66 from JuliaComputing/tb/pr1
litescope: 08072a7      Florent Kermarrec Tue Aug 31 17:48:07 2021 +0200        Copyrights: Bump year.
litespi: 05afb0c        enjoy-digital Sun Oct 24 10:09:15 2021 +0200    Merge pull request #64 from gregdavill/generic_phy_x1_fix
litevideo: 41f3014      Florent Kermarrec Sat Apr 11 19:54:42 2020 +0200        README: switch to markdown.
litex: d36e1b60 enjoy-digital Tue Dec 21 19:22:01 2021 +0100    Merge pull request #1139 from tilk/jtagbone_typo
litex_boards: b8aad4b   enjoy-digital Fri Jan 21 08:05:22 2022 +0100    Merge pull request #332 from tcal-x/prog-cmod-a7
migen: 9a0be7a  occheung  Fri Nov 12 14:04:01 2021 +0800 kasli v2: enable ExploreWithRemap
nmigen: 0b28a97 whitequark Fri Oct 8 17:48:00 2021 +0000 CI: preserve YoWASP cache as well.

```

tcal-x, 2022-01-22:
@bl0x I seem to have reproduced what you saw.   Using the SymbiFlow install from symbiflow-examples, then using `litex_boards/targets/digilent_cmod_a7.py`, the generated bitstream does not seem to work on the CMOD A7 board.   The LEDchaser is not working, and there is no response on ttyUSB1.

bl0x, 2022-01-22:
Updating yosys to 0.13 and arch-defs to the 20211222 version allows to produce a working bitfile.
The main change was to unpin the versions in xc7/environment.yml and run a conda update.

mithro, 2022-01-23:
FYI - @acomodi / @kgugala

acomodi, 2022-01-24:
@bl0x @tcal-x Great to hear that it works. The CMOD A7 should be on the way so I can test on HW as well.

I am not sure though what might have changed in the tools and arch-defs that fixed the issue, so I thing that we might still need to investigate what is going on to have a clearer picture.

tcal-x, 2022-02-10:
@acomodi -- have you received your CMOD A7 yet? :)

acomodi, 2022-02-10:
@tcal-x Yep, actually it has arrived, I am gonna give it a try tomorrow probably

md-raz, 2023-09-21:
Hi All, 

I was wondering if there was any progress on this, as I ran into a similar issue:
I am running the litex_sata_demo example on the Nexys Video Board. All parts of the flow execute without errors, and I get the top.bit file within `\build\nexys_video` folder. Loading the bit file via Vivado hardware manager and openFPGALoader, the bit file is loaded into the flash but with no response from the FPGA over UART or via the LEDs. 

Furthermore, I tried the same flow with the counter example on the Nexys Video board, and also see no response from the LEDs. 

To install F4PGA, I used this guide: https://f4pga-examples.readthedocs.io/en/latest/getting.html
The examples are from here: https://github.com/chipsalliance/f4pga-examples/tree/main/xc7/litex_sata_demo

I appreciate any help. Thanks!