chipsalliance/f4pga-examples #246: ARM Linux can't resolve packages

lachlansmith, 2022-01-21:
I'm on an M1 Mac inside an Ubuntu virutal machine following [SymbiFlow Getting Started](https://symbiflow-examples.readthedocs.io/en/latest/getting-symbiflow.html). When I execute this step:

```
conda env create -f $FPGA_FAM/environment.yml
```

Conda fails to solve the environment most definitely because conda is trying to resolve aarch64 as opposed to x86_64. It fails to resolve the following packages... everything.

```
Collecting package metadata (repodata.json): done
Solving environment: failed

ResolvePackageNotFound: 
  - litex-hub::gcc-riscv64-elf-newlib==9.2.0=20201119_154229
  - litex-hub::symbiflow-yosys-plugins==1.0.0_7_382_g6712d71=20210818_135210
  - litex-hub::vtr-optimized==8.0.0_4118_g06317d042=20210813_070938
  - litex-hub::prjxray-tools==0.1_2842_g6867429c=20210301_104249
  - litex-hub::openocd==0.10.0_1514_ga8edbd020=20201119_154304
  - litex-hub::prjxray-db==0.0_253_gcd41f08=20211122_104637
  - litex-hub::yosys==0.9_5567_g539d4ee9=20210813_070938_py37
```

Is there a work around for this? Surely this issues persists on native linux and somebody has found a work around. Can somebody help me out.

Thanks

pgielda, 2022-01-21:
No real workaround other than building from sources, as you pointed out the pre-compiled packages are available for x86_64. Having said that we will look at possibly providing aarch64 linux packages in the future. The main blocker is that there are no aarch64 github actions runners, we would  have to set up a custom architecture, and previously there was no interest. We could be using QEMU to build but it would potentially take ages to build. So some practical blockers, are solvable if there is interest.

@PiotrZierhoffer @kgugala FYI

lachlansmith, 2022-01-22:
Would be great if Symbiflow could solve this emerging issue. Vivado has no interest as seen here, [Vivado ARM Linux](https://support.xilinx.com/s/question/0D52E00006hprrp/vivado-on-arm-linux?language=en_US).

lachlansmith, 2022-01-22:
Just realised I can give this a try under Rosetta emulation on MacOS. I’ll let you know.

pgielda, 2022-01-24:
There is definitely interest and its definitely doable. Also the whole project is open source so there is no technical obstacle, just a matter of doing it ;) 

I will report once there is some progress on any of the packages. We do need to set-up some CI infra for ARM within the project first to be able to test properly.

lachlansmith, 2022-03-30:
Hi @pgielda understanding these things take time, what does this space look like at the moment?

mithro, 2022-03-30:
FYI - https://github.com/SymbiFlow/conda-packages/pull/160

conversy, 2023-02-27:
Hi, I'm running into this issue too.

I tried to install the `x86_64` version (with the correct installer from miniconda and by specifying an `x86_64` shell with `env /usr/bin/arch -x86_64 /bin/zsh --login` (as per [this](https://developer.apple.com/forums/thread/718666) explanation), to no avail. Also tried with miniforge, still no luck.

Is it possible to see which arch are the modules conda tries to install?

conversy, 2023-03-03:
I tried to setup a debian-based Docker container. I'm running into the same issue, even if the OS is supposedly Linux for aarch64 (`uname`is saying so). Any suggestions?

The Dockerfile:
```
FROM debian

RUN apt update
RUN apt install -y build-essential wget

SHELL ["/bin/bash", "-c"] # for source

COPY f4pga-examples f4pga-examples 
WORKDIR /f4pga-examples

RUN wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-aarch64.sh -O conda_installer.sh

ARG CACHEBUST=1 
RUN echo "$CACHEBUST"

RUN uname
RUN uname -m

RUN ls /f4pga-examples/
ARG F4PGA_INSTALL_DIR=/f4pga-examples/local
ARG FPGA_FAM=xc7
RUN rm -rf local
RUN mkdir -p local/${FPGA_FAM}
RUN bash conda_installer.sh -u -b -p ${F4PGA_INSTALL_DIR}/${FPGA_FAM}/conda

#RUN source "$F4PGA_INSTALL_DIR/$FPGA_FAM/conda/etc/profile.d/conda.sh"
RUN source "local/$FPGA_FAM/conda/etc/profile.d/conda.sh" && \
    conda env create -f $FPGA_FAM/environment.yml

```

unixb0y, 2023-05-12:
> No real workaround other than building from sources
Are there any instructions to build all required dependencies from source?