chipsalliance/f4pga-examples #263: Symbiflow-classroom: Array Initializers  Do Not Work

nelsobe, 2022-03-08:
SYMBIFLOW-CLASSROOM-PROJECT

If SV code contains a legal array declaration and initialization like this:
`
      // Lookup table for % 3 calculation.
    logic[1:0] modmem[8] = '{0, 1, 2, 0, 1, 2, 0, 1};
`
a syntax error results.   Arrays cannot be initialized this way with the tools (but it is legal SV and works with other tools).

nelsobe, 2022-03-09:
@mithro @acomodi @tmichalak @mkurc-ant Would be interested in any feedback you can provide.

rkapuscik, 2022-03-09:
Could you provide more details on which tool are you using and with what options? I verified that UHDM plugin for yosys correctly parses this declaration - it is present in UHDM and in Yosys AST. I tested this using both Surelog to create UHDM file and `read_verilog_with_uhdm` in yosys with plugin.

On the other hand, using this in original yosys' verilog frontend does cause a syntax error:
```
yosys> read_verilog -sv tests/top.sv
1. Executing Verilog-2005 frontend: tests/top.sv
Parsing SystemVerilog input from `tests/top.sv' to AST representation.
tests/IncompleteTop/top.sv:6: ERROR: syntax error, unexpected OP_CAST
```

nelsobe, 2022-03-09:
@rkapuscik Thanks for the reply.  All of issues #263 through #268 occur using the original yosys verilog frontend.

jcrob2, 2022-06-17:
Using the following timestamp and hash, the issue still exists.
Timestamp: 20220606-234655
Hash: 1667c14

I've included a file that contains a list of the content/versions of the conda environment that we have.
[conda_list.txt](https://github.com/chipsalliance/f4pga-examples/files/8931052/conda_list.txt)