chipsalliance/f4pga-examples #264: Symbiflow-classroom: "Error declaring a variable inside an always_comb block"

nelsobe, 2022-03-09:
SYMBIFLOW-CLASSROOM-PROJECT

The tools give compile errors when the integer is declared local to the always_comb body:

`
always_comb begin
    integer index;

    if (hcount < H_PIXELS && vcount < V_PIXELS) begin
        for (index = 0; index < 4; index = index + 1) begin
            VGA_R[index] = rd_data[2];
            VGA_G[index] = rd_data[1];
            VGA_B[index] = rd_data[0];
        end
    end else begin
        VGA_R = 4'b0000;
        VGA_B = 4'b0000;
        VGA_G = 4'b0000;
    end
end
`

nelsobe, 2022-03-09:
@mithro @acomodi @tmichalak @mkurc-ant Would be interested in any feedback you can provide.

tmichalak, 2022-03-17:
@nelsobe could you provide more details on the errors you get?

nelsobe, 2022-03-22:
@tmichalak Working on getting a trace from the student.

nelsobe, 2022-03-28:
@tmichalak The error provided by the tools is that a latch is inferred on the "index" signal and the compilation fails.

We do not believe that a latch should be inferred (or the compilation fail) since "index" is local to the always_comb  block.   In this code "index" is functioning like a VHDL variable - used to help in the computation but not intended to carry information out of the block.