chipsalliance/f4pga-examples #265: Symbiflow-classroom: "Multiple include causes error"

nelsobe, 2022-03-09:
SYMBIFLOW-CLASSROOM-PROJECT

Using the include macro to include the same module more than once in a project (perhaps from different locations in the files) causes an error.  While bad practice, this causes an error for the tools while Vivado just redefines the module.

nelsobe, 2022-03-09:
@mithro @acomodi @tmichalak @mkurc-ant Would be interested in any feedback you can provide.

rkapuscik, 2022-03-09:
Could you specify the tool you're using and the exact error?

I've created [this test](https://github.com/chipsalliance/UHDM-integration-tests/pull/686). It does fail in Yosys with Verilog frontend, but passes in UHDM frontend. I tested it also in Surelog and Verilator and they issue only a warning in this case.