chipsalliance/f4pga-examples #266: Symbiflow-classroom: Older style parameter declarations not supported

nelsobe, 2022-03-09:
In SystemVerilog a parameter to a module can be define in one of two ways:

`
module riscv_mem #(parameter TEXT_MEMORY_FILENAME) (clk, rst, PC, , ...);
`
as well as:
`
module riscv_mem (clk, rst, PC, , ...);
    parameter TEXT_MEMORY_FILENAME;
`

The first way (the newer way) is not supported.

nelsobe, 2022-03-09:
@mithro @acomodi @tmichalak @mkurc-ant Would be interested in any feedback you can provide.

rkapuscik, 2022-03-09:
Could you specify which tool are you trying to run and with what options? I.e. for Yosys with UHDM plugin, we have tests that cover both [first](https://github.com/chipsalliance/UHDM-integration-tests/blob/master/tests/ReplicationPassedToInstance/top.sv) and [second](https://github.com/chipsalliance/UHDM-integration-tests/blob/master/tests/ParameterFromPkgInNestedModule/top.sv) way and both of them are supported.

If you could also provide the error output that you're seeing and a minimal complete example to reproduce, that would be really helpful.

nelsobe, 2022-03-09:
Using original Yosys front end.

nelsobe, 2022-03-17:
@rkapuscik Can you clarify something for us?  In our mind there are two ways to run the toolchain.  The first is the "out of the box" install which we have interpreted as being the Yosys front end (thus my answer above).  The other way we understand is Surelog+UHDM front end.  You mention Yosys with UHDM, is this a third way or is this the first way I mentioned in my response?  Thanks.

rkapuscik, 2022-03-18:
You're right. There are two main ways - original Yosys Verilog frontend and UHDM plugin frontend. Yosys with UHDM plugin is the same as Yosys with Surelog+UHDM.

One detail here is that the plugin enables two flows: 
1.  reading UHDM file created separately by using Surelog (`read_uhdm` command) 
2.  reading Verilog and SystemVerilog files directly, using Surelog inside the plugin as a library (and not creating a separate UHDM file, just an in-memory representation). This is done by `read_verilog_with_uhdm` command.

[image]

As a side note, UHDM is just an implementation detail and we are renaming the command in 2. to just `read_systemverilog` (see https://github.com/SymbiFlow/yosys-f4pga-plugins/pull/270), but both of the commands above will be kept for backwards compatibility.

nelsobe, 2022-04-14:
I have done some re-testing and this does seem to work with Yosys.  I don't know if it was because the distribution was behind on Yosys (which we discovered with other issues), maybe that was the issue.  But it is working now so, I believe it can be closed.

closed 2022-04-14 by nelsobe