chipsalliance/f4pga-examples #267: Symbiflow-classroom: Parameterizing top-level module

nelsobe, 2022-03-09:
SYMBIFLOW-CLASSROOM-PROJECT

If a top-level module in a design has a parameter associated with it, is there a way to provide a value for that parameter when the tool flow is run?

nelsobe, 2022-03-09:
@mithro @acomodi @tmichalak @mkurc-ant Would be interested in any feedback you can provide.

rkapuscik, 2022-03-09:
Please specify the tool. For:
* Yosys with UHDM plugin: `-Pparameter=value` overrides a toplevel module parameter
  * using Surelog separately: add it to the command, i.e. `surelog -parse -Pparameter=value top.sv`, then use `read_uhdm` in Yosys
  * using UHDM frontend directly (which uses Surelog under the hood): add it to the `read_verilog_with_uhdm` command, i.e. `read_verilog_with_uhdm -parse -Pparameter=value top.sv`
* Yosys Verilog frontend: [this issue](https://github.com/YosysHQ/yosys/issues/132) suggests to use `chparam` in Yosys script

Please let me know if you have any further questions.

nelsobe, 2022-03-17:
@rkapuscik Thanks for the info - very helpful!

closed 2022-03-17 by nelsobe