chipsalliance/f4pga-examples #268: Symbiflow-classroom: MMCM placement error

nelsobe, 2022-03-09:
SYMBIFLOW-CLASSROOM-PROJECT

Compiling the included files results in the following error.  The design has 2 MMCM's - related to this?

```
(xc7) nelson@CB461-EE11591:~/lab07_io_clocks$ make
cd /home/nelson/lab07_io_clocks/build/basys3 && symbiflow_synth -t io_clocks -v /home/nelson/lab07_io_clocks/io_clocks.sv -d artix7 -p xc7a35tcpg236-1 -x /home/nelson/lab07_io_clocks/Basys3_Master.xdc 2>&1 > /dev/null
cd /home/nelson/lab07_io_clocks/build/basys3 && symbiflow_pack -e io_clocks.eblif -d xc7a50t_test  2>&1 > /dev/null
cd /home/nelson/lab07_io_clocks/build/basys3 && symbiflow_place -e io_clocks.eblif -d xc7a50t_test  -n io_clocks.net -P xc7a35tcpg236-1  2>&1 > /dev/null
Traceback (most recent call last):
  File "/home/nelson/opt/f4pga/xc7/install/share/symbiflow/scripts/prjxray_create_place_constraints.py", line 1036, in <module>
    main()
  File "/home/nelson/opt/f4pga/xc7/install/share/symbiflow/scripts/prjxray_create_place_constraints.py", line 961, in main
    blocks, grid_capacities):
  File "/home/nelson/opt/f4pga/xc7/install/share/symbiflow/scripts/prjxray_create_place_constraints.py", line 680, in place_clocks
    assert loc is not None, (clock_name, available_placements[key])
AssertionError: ('mmcm_vga', ['MMCME2_ADV_X1Y0'])
make: *** [/home/nelson/f4pga-examples/common/common.mk:62: /home/nelson/lab07_io_clocks/build/basys3/io_clocks.place] Error 1
```
[mmcm.zip](https://github.com/chipsalliance/f4pga-examples/files/8211043/mmcm.zip)

nelsobe, 2022-03-09:
@mithro @acomodi @tmichalak @mkurc-ant Would be interested in any feedback you can provide.

mkurc-ant, 2022-03-11:
Let me see what the problem might be.

mkurc-ant, 2022-03-11:
@nelsobe The problem was with the clock constraint generator script which was constraining both `MMCM`s to the same site. This was due to the hard requirement of having a `PLL`/`MMCM` and an `IBUF` that drives it in the same CMT region.

I have created a PR with a fix: https://github.com/SymbiFlow/f4pga-arch-defs/pull/2493

mkurc-ant, 2022-03-18:
@nelsobe I think I have finally solved the placement problem. I added a test for dual MMCM design as well.

One note though: You need to set `COMPENSATION` to `ZHOLD` in your design or remove `BUFG`s from the feedback loop. This is because by default compensation is set to `INTERNAL` which implies internal `MMCM` feedback. Currently the toolchain cannot infer the setting basing on the actual connectivity as Vivado does.

mkurc-ant, 2022-03-29:
@nelsobe The PR with the fix has been merged. Please verify that the problem is gone.

mkurc-ant, 2022-04-05:
@nelsobe Right. The problem was that apart from merging the fix into `f4pga-arch-defs` we also need the CI to generate an installable package and upload it to GCP.

That mechanism was broken. We have fixed it recently and the new package should be available as of now. There is an open PR to `f4pga-examples` which bumps its version: https://github.com/chipsalliance/f4pga-examples/pull/282. You may try installing examples from that PR.

mkurc-ant, 2022-04-05:
The PR is meged now and the current version of `f4pga-examples` should include the fix on the master branch

nelsobe, 2022-04-14:
My testing shows it now working.  I believe this issue can be closed.