chipsalliance/f4pga-examples #269: Symbiflow-classroom: Surelog/UHDM mishandles multi-bit input signals

nelsobe, 2022-03-09:
SYMBIFLOW-CLASSROOM-PROJECT

Using Surelog/UHDM front end

Attached is a simple design that compiles successfully using 
yosys standard frontend. All it does is turn on LEDs when switches are flipped.

This design compiles successfully using SURELOG/UHDM if, instead of naming the 
input logic ports iteritavely (sw[0],sw[1],sw[2]), 
they are named uniquely (A,B,C).

The issue is shown in compile.log that by the time that the design reaches
Yosys AST, the switch ports have already been flattened which suggests that the
issue resides in Surelog or UHDM.

[SwitchBug.zip](https://github.com/chipsalliance/f4pga-examples/files/8217739/SwitchBug.zip)

kbieganski, 2022-03-11:
It does indeed seem to be a bug in Surelog. Filed an issue: https://github.com/chipsalliance/Surelog/issues/2733

rkapuscik, 2022-03-18:
The Surelog issue was resolved, but we need to update the plugin to use the newest version. We'll let you know when it's done.

rkapuscik, 2022-03-31:
Sorry that I wasn't clear enough. By updating Surelog, I mean that

1. New version of Surelog is available on [litex-hub](https://anaconda.org/LiteX-Hub/repo) (it is used by conda to setup the tools)
2. We verify that the new version works with our plugin (tracked in the [integration repository](https://github.com/antmicro/yosys-uhdm-plugin-integration))

Reinstalling f4pga-examples will give you the updated version, but you can also update just the conda environment:
```
conda env update -f xc7/environment.yml
```
If anything goes wrong, you can also remove and recreate it by
```
conda env remove -n xc7
conda env create -f xc7/environment.yml
```
This should give you the updated tools. You can verify which version you have installed by 
```
conda env export -n xc7
```

nelsobe, 2022-04-15:
@rkapuscik Thanks!  Have verified it works, closing issue.

closed 2022-04-15 by nelsobe